carla-simulator/carla #1092: several questions about ver 0.9x

coreper, 2019-01-03:
I have several question for the new Carla API:
1. Are there any methods to view all the starting points on a map? If I want to start the vehicle actor at the roundabout, how can I setup the starting position?

2. If I want to add some noise on the auto pilot steer control, how can I do it?

3. How can I know how much the vehicle is off the road such as version 0.8x? The new lane detector seems only detect the lane type (solid or broken).

anshulpaigwar, 2019-01-03:
To find the starting points check this thread: [link](https://github.com/carla-simulator/scenario_runner/issues/18)

Another pretty simple way I used to find the coordinates for my desired location in the Carla world just launch the python script `manual_control.py` and you see the coordinates of current location of the vehicle in the left panel of pygame window. 
Now all you have to do is to drive your vehicle to desired location :))

coreper, 2019-01-03:
> To find the starting points check this thread: [link](https://github.com/carla-simulator/scenario_runner/issues/18)
>
> Another pretty simple way I used to find the coordinates for my desired location [...]

Thank you! That was smart!

marcgpuig, 2019-01-14:
Hi @coreper 

> How can I know how much the vehicle is off the road such as version 0.8x? The new lane detector seems only detect the lane type (solid or broken).

You can get a waypoint in your **car position**. This waypoint will be **centered** on the nearest lane, so you can get the distance of these points :)

Cheers

closed 2019-03-22 by stale[bot]