carla-simulator/carla #1095: WARNING: SpawnActor: failed with: rpc::rpc_error during call

JackRen88, 2019-01-04:
Hello,everyone!
I want to see where the origin of the world in carla is，so I modify vehicle's spawn point as carla.Transform() in the manual.py,when I run the script,showing as follow:
`WARNING: SpawnActor: failed with: rpc::rpc_error during call`,and resulting in pygame not starting properly.How should I solve it?

Sakrust, 2019-01-04:
The origin of the Carla Map is inside the fountain of the roundabout. At (0, 0, 0) you cannot spawn a vehicle. Try (0, 0, 20) and the vehicle will fall down from above or choose a location near to (0, 0) which is at ground level.

nsubiron, 2019-01-15:
The usual cause of this message is a collision at the spawn point, it can be because there is another, a building, or any other obstacle.

The main problem here is the lack of information, the spawn function actually returns a string with the cause of failure but there is a problem retrieving this message on the client-side. I'll add this issue to our backlog to track this problem.

closed 2019-01-29 by nsubiron