carla-simulator/carla #1096: Python API Documentation

sagnibak, 2019-01-05:
Am I missing something or does documentation does not exist for the Python API for versions 0.8.x? I don't see any on the official CARLA docs, and neither did I find any in the prebuilt install for Windows. Will I have to build from source in order to also build the documentation?

fabianoboril, 2019-01-06:
Hi,
all documentation in 0.8.4 can be found here:
https://github.com/carla-simulator/carla/tree/0.8.4/Docs
As you can see, there is no detailed documentation on the Python API used in 0.8.4. Yet, you can look into the examples in the PythonClient/ folder to understand the Python commands.

sagnibak, 2019-01-06:
I did exactly that 😅 but I was wondering if they had a better way (I guess not)

closed 2019-03-17 by stale[bot]