carla-simulator/carla #1100: Problem understanding the tutorial

geekhut, 2019-01-07:
So I'm reading this tutorial on how to make a new map in CARLA with RoadRunner:

https://github.com/carla-simulator/carla/blob/master/Docs/how_to_make_a_new_map.md#working-with-maps-in-unreal

I am halfway through and kinda confused with **Step 2 - Unreal -> Working with maps in Unreal:**
It's kinda vague that it says "First of all create a level with the Map name (this will be important later) and save it inside Game/Carla/Maps/
Once everything is loaded up you should end with several staticMeshes in the folder you have selected. Drag them all to the level and you will have your map in Unreal! Congratulations!."

How should I go about doing this?

P.s. I am completely new to this CARLA thing, so far I have only tried running the precompiled simulation of CARLA, and my boss wants me to try building roads from scratch using RoadRunner...

XGodina, 2019-01-07:
Hi @geekhut 
I recommend that you download the free trial RoadRunner in https://www.vectorzero.io/ and to try it a little. You can watch the tutorials in youtube. If you do not know the Unreal I recommend the same.

About the Step 2 - We mean that the name of the opendrive and the map Should are the same so that no problems arise.

Unlingius, 2019-01-09:
@geekhut    maybe it will help  https://github.com/carla-simulator/carla/issues/1015

closed 2019-01-15 by marcgpuig