carla-simulator/carla #1101: Is it possible to run carla simulator binary in ubuntu 14.04?

tanglrHello, 2019-01-07:
I have a package built on ubuntu16.04, and have a server which is of version ubuntu14.04. I want to run the carla simulator server by ./CarlaUE4.sh without display on the 14.04 server.
I copied the directory from Dist to the 14.04 server, and execute:
$DISPLAY= ./CarlaUE4.sh 
Then a segmentation fault comes

My configuration:
ubuntu 14.04
carla release 0.9.1
NVIDIA GEFORCE GTX 1080Ti
nvidia driver: 384.130

I know there is a note in document saying carla requires ubuntu16.04. But if I don't need to compile it on ubuntu 14.04 and just run a prebuilt binary, is it still necessary? As updating the os version for the server or preparing a docker is a little bit toublesome, is there a way to just install something to get it right?

Any suggestion or clue is really aprreciated. If you have run the carla simulator successfully on ubuntu 14.04, please let me know. Thanks!

The complete log for the segmentation fault is here:

> 
Increasing per-process limit of core file size to infinity.
LogPlatformFile: Using cached read wrapper
LogInit: Display: RandInit(1949752523) SRandInit(1949752523).
LogInit: Overriding engine version to CL 0, compatible CL 0
LogTaskGraph: Started task graph with 5 named threads and 17 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.036181
LogICUInternationalization: Display: ICU TimeZone Detection - Raw Offset: +8:00, Platform Override: ''
LogPluginManager: Mounting plugin Carla
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin RSMapGen
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin PerformanceMonitor
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin CryptoKeys
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin DatasmithContent
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin ImgMedia
LogPluginManager: Mounting plugin LinearTimecode
LogPluginManager: Mounting plugin MediaCompositing
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin ActorSequence
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogStreaming: Display: Took  0.170s to configure plugins.
LogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file 
LogInit: Using libcurl 7.57.0
LogInit:  - built for x86_64-pc-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_TLSAUTH_SRP
LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
[... 208 lines cut ...]
[2019.01.07-08.42.35:107][  0]LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
[2019.01.07-08.42.35:108][  0]LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
[2019.01.07-08.42.35:108][  0]LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
[2019.01.07-08.42.35:108][  0]LogInit:  - MaxHostConnections = 16  - Libcurl will limit the number of connections to a host
[2019.01.07-08.42.35:108][  0]LogInit:  - LocalHostAddr = Default
[2019.01.07-08.42.35:108][  0]CrashReportClientLog: Sending HTTP request: https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data?AppID=CrashReporter&AppVersion=4.19.2-0%2B%2B%2BUE4%2BRelease-4.19&AppEnvironment=Release&UploadType=crashreports&UserID=0002a8b825010d07003b03112d9280bf%7C%7C
[2019.01.07-08.42.36:570][  0]CrashReportClientLog: OnProcessRequestComplete(), State=SendingFiles bSucceeded=1
[2019.01.07-08.42.36:570][  0]CrashReportClientLog: State change from SendingFiles to SendingFiles
[2019.01.07-08.42.36:570][  0]CrashReportClientLog: All uploads done
[2019.01.07-08.42.36:570][  0]CrashReportClientLog: State change from SendingFiles to Finished
[2019.01.07-08.42.37:105][  0]LogLinux: FLinuxPlatformMisc::RequestExit(0)
[2019.01.07-08.42.37:105][  0]LogGenericPlatformMisc: FPlatformMisc::RequestExit(0)
[2019.01.07-08.42.37:138][  0]CrashReportClientLog: Final state (Receiver) = Finished
[2019.01.07-08.42.37:138][  0]CrashReportClientLog: Final state (Receiver) = Unknown UploadState value
[2019.01.07-08.42.37:138][  0]LogExit: Preparing to exit.
[2019.01.07-08.42.37:139][  0]LogExit: Object subsystem successfully closed.
[2019.01.07-08.42.37:141][  0]LogModuleManager: Shutting down and abandoning module HTTP (10)
[2019.01.07-08.42.37:159][  0]LogModuleManager: Shutting down and abandoning module SSL (9)
[2019.01.07-08.42.37:159][  0]LogModuleManager: Shutting down and abandoning module CrashDebugHelper (6)
[2019.01.07-08.42.37:159][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (4)
[2019.01.07-08.42.37:159][  0]LogModuleManager: Shutting down and abandoning module SandboxFile (2)
[2019.01.07-08.42.37:160][  0]LogExit: Exiting.
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault
>

pawel-ziecina, 2019-01-07:
Error `Linux_PlatformCreateDummyGLWindow` is often related with inproper run on headless machines.

Your start command (`$DISPLAY= ./CarlaUE4.sh`) seems to not follow [documentation](https://carla.readthedocs.io/en/latest/carla_headless/).

tanglrHello, 2019-01-08:
@pawel-ziecina 
Thank you for your reply! In fact I was following the documentation here: https://carla.readthedocs.io/en/latest/configuring_the_simulation/#running-off-screen

Is there a big difference between the two methods? To be honest, I was a little confused when I saw the two different instructions. But as the one I posted seems to be simpler for me to achieve my purpose, I followed it without manually installing anything mentioned in https://carla.readthedocs.io/en/latest/carla_headless/

By the way, I'd like to know whether headless exactly means running without display? Or headless is something more than that?

pawel-ziecina, 2019-01-08:
Ah - you are right, just checked this on 16.04 and your method worked.

I've observed same error (with opengl initialization) while trying to run carla in docker on GKE, so thought that could be the same issue. Cause of my problems was lack of nvidia egl dynamic library in LD_LIBRARY_PATH. You can check it with:

- check name of nvidia egl library with `cat /usr/share/glvnd/egl_vendor.d/*nvidia*`
- check if library exists in `ldconfig -v|grep -i egl`

I understand headless workstation as computer without X server. But not sure if my understanding is correct :)

tanglrHello, 2019-01-08:
@pawel-ziecina 
Maybe this is the problem for me ^_^
The server which I can't run carla on seems not to have a valid egl. There is even not /usr/share/glvnd directory. And I checked my local machine, it has the directory for egl. 

And with ldconfig,  I got  the following infomation on the wrong machine:

/sbin/ldconfig.real: Path `/lib/x86_64-linux-gnu' given more than once
/sbin/ldconfig.real: Path `/usr/lib/x86_64-linux-gnu' given more than once
/sbin/ldconfig.real: /lib/x86_64-linux-gnu/ld-2.23.so is the dynamic linker, ignoring

/sbin/ldconfig.real: Cannot stat /usr/lib/x86_64-linux-gnu/libxlutil.so: No such file or directory
	libwayland-egl.so.1 -> libwayland-egl.so.1.0.0
	libQt5EglDeviceIntegration.so.5 -> libQt5EglDeviceIntegration.so.5.5.1
	libhybris-eglplatformcommon.so.1 -> libhybris-eglplatformcommon.so.1.0.0
/sbin/ldconfig.real: /lib32/ld-2.23.so is the dynamic linker, ignoring

/usr/lib/x86_64-linux-gnu/mesa-egl:
	libEGL.so.1 -> libEGL.so.1.0.0
/sbin/ldconfig.real: Cannot stat /usr/lib/libopen-rte.so: No such file or directory
/sbin/ldconfig.real: Cannot stat /usr/lib/liboshmem.so: No such file or directory
/sbin/ldconfig.real: Cannot stat /usr/lib/libmca_common_sm.so: No such file or directory
/sbin/ldconfig.real: Cannot stat /usr/lib/libopen-pal.so: No such file or directory
/sbin/ldconfig.real: Can't create temporary cache file /etc/ld.so.cache~: Permission denied

So should I add the installation for egl into Dockerfile? Or install everything mentioned in https://carla.readthedocs.io/en/latest/carla_headless/ into the docker image? How did you solve it? Hope for some shortcut to the solution :)

pawel-ziecina, 2019-01-08:
Switching to newer drivers might help - not sure if using those from /var/lib/nvidia-docker. I think EGL was added in some version of nvidia drivers - not sure which.

In my case I've added required libraries to LD_LIBRARY_PATH with:

```
sudo echo /usr/local/nvidia/lib64 >> /etc/ld.so.conf.d/glvnd.conf
sudo ldconfig
```

tanglrHello, 2019-01-08:
The nvidia driver is newer on the server than that on my local system. The driver version for local system is  384.130, and for the server is 390.59. But I can't find libEGL.so.390.59 on the server. It seems the nvidia driver was installed in some unusual way. Thank you so much anyway for your patient help :) @pawel-ziecina

tanglrHello, 2019-01-11:
After struggling with this problem for days, finally I got sudo privilege on one of our headless server. I followed the instruction in  https://carla.readthedocs.io/en/latest/carla_headless/ and install everything mentioned. And glxinfo can be run smoothly in the documented way. Then I can run CarlaUE4 on the headless server without signal 11 caught. By the way, the headless server I have sudo privilege on is of ubuntu 16.04, not sure about the result on ubuntu14.04.

But though the segment fault disappeared, the CarlaUE4 seems to stuck after Initializing movie player. The next log is expected to be:
[2019.01.11-03.06.14:787][  0]LogUObjectArray: 32687 objects as part of root set at end of initial load.
[2019.01.11-03.06.14:787][  0]LogUObjectAllocator: 5984496 out of 0 bytes used by permanent object pool.
[2019.01.11-03.06.14:787][  0]LogUObjectArray: CloseDisregardForGC: 0/0 objects in disregard for GC pool
[2019.01.11-03.06.14:787][  0]LogEngine: Initializing Engine...
But it just got stuck. The graphic card on the headless server is Geforce gtx 1080Ti and driver version is 396.44. Till now the carla server has not been set up successfully and can't see port 2000 be occupied by CarlaUE4. With nvidia-smi, I can see CarlaUE4 is running with GPU0:
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0     16902      G   ./CarlaUE4/Binaries/Linux/CarlaUE4           143MiB |
|    0     31276      G   /usr/lib/xorg/Xorg                            11MiB |
|    1     31276      G   /usr/lib/xorg/Xorg                            10MiB |
|    2     31276      G   /usr/lib/xorg/Xorg                            10MiB |
|    3     31276      G   /usr/lib/xorg/Xorg                            10MiB |
|    4     31276      G   /usr/lib/xorg/Xorg                            10MiB |
|    5     31276      G   /usr/lib/xorg/Xorg                            10MiB |
|    6     31276      G   /usr/lib/xorg/Xorg                            10MiB |
|    7     31276      G   /usr/lib/xorg/Xorg                            10MiB |
+-----------------------------------------------------------------------------+

closed 2019-03-19 by stale[bot]