carla-simulator/carla #1103: Add vehicle blueprint

Sakrust, 2019-01-07:
I created my own vehicle blueprint and added a folder inside Content/Carla/Blueprints/Vehicles, then added this blueprint value to VehicleSpawner and VehicleFactory blueprints.

If I now try to spawn my custom vehicle via PythonAPI, the new blueprint is missing in the blueprint library (world.get_blueprint_library()). Do I miss a step? How to get access to my blueprint?

Using Carla 0.9.1 on Windows.

Sakrust, 2019-01-09:
Update: If my created vehicle is of class CarlaWheeledVehicle, I have no problem adding and spawning it via python. But if i want to use a Vehicle of a different class, how can I add this vehicle to Carla?

TheNihilisticRobot, 2019-01-15:
Hi @Sakrust 

Is there any specific reason you don't want your vehicle to be a Child of CarlaWheeledVehicle? If not, you could take your own created vehicle and make it child of CarlaWheeledVehicle by opening the ClassSettings tab in your vehicle's blueprint editor and changing its parent class.

Sakrust, 2019-01-18:
The reason is, I want to bypass the engine simulation part of the WheeledVehicle class.
 Therefore I need to change my WheeledVehicleMovementComponent to SimpleWheeledVehicleMovementComponent. But i cannot change this MovementComponent for all CarlaWheeledVehicles, so I created a new VehicleClass.

Sakrust, 2019-01-21:
### Short Update on this issue:

I created a new class inheriting from CarlaWheeledVehicle. I can add this Vehicle to the VehicleFactory blueprint  and this vehicle is found via world.get_blueprint_library().

> ActorBlueprint(id=vehicle.simple.audi,tags=[simple, vehicle, audi])

But if I try to spawn this vehicle I get an Error: 

> LogCarla: Warning: Failed to spawn actor 'vehicle.simple.audi'
> WARNING: SpawnActor: failed with: rpc::rpc_error during call 
> RuntimeError: rpc::rpc_error during call in function spawn_actor

The client script stops, but in the simulator view you can see the spawned vehicle.

Any ideas on how to maybe register the actor properly?

Second issue:

If I place my vehicle anywhere in the world in editor mode and rebuild Carla, I can see my vehicle in the simulator view, but it does not appear in the actor list (world.get_actors())

Do I need to add placed actors anywhere so they will be recognized?

nsubiron, 2019-01-23:
Hi @Sakrust,

I can add a couple of comments from the programming side.

> RuntimeError: rpc::rpc_error during call in function spawn_actor

Check out the simulator output for this, there is currently a problem retrieving the error message on the client side.

> If I place my vehicle anywhere in the world in editor mode and rebuild Carla, I can see my vehicle in the simulator view, but it does not appear in the actor list (world.get_actors())

The actors need to be created with out spawn system otherwise they're not added to our actor registry. At this point you can only do this from the client, we want to add this functionality inside Unreal too but it's not yet implemented.

> Do I need to add placed actors anywhere so they will be recognized?

What you can do as workaround is what we do for the traffic lights [here](https://github.com/carla-simulator/carla/blob/cdd3c83ba0a2061ef76058e6f16638e48697aa08/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEpisode.cpp#L91-L99). At begin play we parse the actors of a certain type and add them to the registry.

Sakrust, 2019-01-28:
Finally found a solution. Replicated a BaseVehiclePawn for my vehicle class and created a new VehicleFactory which also has to be added to GameMode. Thank you for your hints.

closed 2019-01-28 by Sakrust

YDSh510, 2019-02-20:
Hi @Sakrust ,

I have to add a new vehicle model in carla simulation . need your support as new to It.

I just have .fbx file how can I proceed further . any video you have as I am unable to understand the steps given in documenation.

Sakrust, 2019-02-20:
I never added a new mesh or something like that, always used the provided assets, just modified the behaviour of the vehicles. I'm afraid I won't be able to help you.

YDSh510, 2019-02-20:
ok thanks

WaelKarkoub, 2019-03-18:
> Update: If my created vehicle is of class CarlaWheeledVehicle, I have no problem adding and spawning it via python. But if i want to use a Vehicle of a different class, how can I add this vehicle to Carla?

Hey do you mind showing me how you made your Vehicle spawnable with python? Im currently struggling with that. Thank you

Sakrust, 2019-03-18:
I've added a MyBaseVehiclePawn of my class, adapted all blueprint code and parameter, and then created a vehicle factory blueprint for my class, which needs to be added to your GameMode blueprint. 

After these steps you should be able to create new vehicle blueprints deriving from your MyBaseVehiclePawn. 

You need to copy and adapt all necessary functionality from CarlaWheeledVehicle.h and related classes to your vehicles and thus be able to spawn your vehicle via Python.

stevendae, 2019-03-24:
@Sakrust I would like to spawn a new asset I added that I adapted from the LincolnMK2017.

I created the blueprint and followed the documentation for configuring the new blueprint. I then added the blueprint to the 4WheeledVehicles list in the VehicleSpawner blueprint. But I can't spawn it when I run a PythonAPI script. Do you know what I'm missing?

stevendae, 2019-03-24:
Also when I try and save the CarlaGameMode blueprint I'm getting notifications saying that it contains invalid data. Any ideas?

a-jahani, 2019-11-26:
> we want to add this functionality inside Unreal too but it's not yet implemented.

If I'm not mistaken, I'm pretty sure this functionality was in UE4. You just have to make sure the actor is replicated through the network.

nsubiron, 2019-11-27:
Hi @a-jahani,
This is our own registry system for actors, independent of UE4, this way we can keep only the actors that are relevant for our simulation. What I meant is that we wanted to add methods to add actors to the Carla registry from within Unreal Editor. Still at this point it's only possible from the client-side.

Akshay-sudo-ux, 2020-03-13:
Hi @Sakrust 
Can you share any reference document from where you got the above steps.

hubing1994, 2020-04-29:
> Hi @a-jahani,
> This is our own registry system for actors, independent of UE4, this way we can keep only the actors that are relevant for our simulation. What I meant is that we wanted to add methods [...]

Hi @Néstor Subirón,
How is it going now？ I want to spawn some cars in ue4 editor, but world.get_actors() get nothing in client.

mindThomas, 2020-06-12:
@Sakrust Does that mean that you managed to get a car blueprint based on the SimpleWheeledVehicleMovementComponent working with CARLA?
Did that improve the physics and especially of the car? Is it closer to actual driving rather than the fairly crappy vehicle dynamics coming with the stock CARLA cars (CarlaWheeledVehicle)?
If so, would you mind to share your blueprint and the exact lines of code you've changed to get it spawned and connected with CARLA?
Thank you in advance.