carla-simulator/carla #1104: Loosing the wheels

pawel-ziecina, 2019-01-07:
Hi, below is method for "unscreewing the wheels" in most (all?) vehicles.

```python
import carla
import random
c = carla.Client('localhost', 2000)
w = c.get_world()
vbs = w.get_blueprint_library().filter('vehicle*')
vb = random.choice(vbs)
sp = w.get_map().get_spawn_points()[0]
v = w.spawn_actor(vb, sp)
sp.rotation.pitch = -90
sp.location.z = 40
w.get_spectator().set_transform(sp)
v.set_simulate_physics(False)
v.apply_control(carla.VehicleControl(throttle=10.0))
v.set_simulate_physics(True)
```

OS: Ubuntu 18.03
Carla: 0.9.2 from official release

pawel-ziecina, 2019-01-07:
It was just play with API. I don't claim it is logic/common sequence of operations.

You are right when disabled physics before appling control to car it wouldn't move.

But when you enable physics again - car body will move, while wheels will stay in place. I think it is a bug.

[image]

tanglrHello, 2019-01-08:
This seems to be the same problem I reported in #1001 . It has been fixed in release 0.9.2.

pawel-ziecina, 2019-01-08:
You are right - it is the same issue.

Confirming that I'm using official 0.9.2 build.

[image]

nsubiron, 2019-01-11:
I think it is related but different than #1001, only some vehicles were loosing the wheels and it was due a corruption of their physics asset in UE4. That was already fixed in 0.9.2. Not sure what's causing this, we should investigate.

pawel-ziecina, 2019-03-15:
Could not reproduce on 0.9.4.

closed 2019-03-15 by pawel-ziecina