carla-simulator/carla #1106: Not consequentive frames in on_tick callback

pawel-ziecina, 2019-01-07:
Hi, I'm observing not consecutive frame numbers in `on_tick` callback.

As in below example: 77369 -> 77371 -> 77370 -> 77372

```
Not consequentive timestamps prev=Timestamp(frame_count=77369,elapsed_seconds=1372.1,delta_seconds=0.0328785,platform_timestamp=1.66315e+06) current=Timestamp(frame_count=77371,elapsed_seconds=1372.16,delta_seconds=0.0654068,platform_timestamp=1.66315e+06)
Not consequentive timestamps prev=Timestamp(frame_count=77371,elapsed_seconds=1372.16,delta_seconds=0.0654068,platform_timestamp=1.66315e+06) current=Timestamp(frame_count=77370,elapsed_seconds=1372.13,delta_seconds=-0.0327082,platform_timestamp=1.66315e+06)
Not consequentive timestamps prev=Timestamp(frame_count=77370,elapsed_seconds=1372.13,delta_seconds=-0.0327082,platform_timestamp=1.66315e+06) current=Timestamp(frame_count=77372,elapsed_seconds=1372.19,delta_seconds=0.0655386,platform_timestamp=1.66315e+06)
```

Is it known/intended behaviour? Code to reproduce.

```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-
from threading import Lock
import time
import carla

last_frame = None
lock = Lock()

def on_tick(timestamp):
    global last_frame
    global lock
    with lock:
        if last_frame and last_frame.frame_count + 1 != timestamp.frame_count:
            print(f'Not consequentive timestamps prev={last_frame} current={timestamp}')
        last_frame = timestamp

client = carla.Client('localhost', 2000)
client.set_timeout(2.0)
world = client.get_world()
world.on_tick(on_tick)
time.sleep(60)
```

fabianoboril, 2019-01-07:
Hi,
the on_tick method is a callback, which means that it is executed in a separate thread on every call. Because of that, it can happen that several callbacks are executed "concurrently".  The lock only helps that the threads are executed on after another, but not that an "old" thread is executed always before a "new" one. I think, currently, the only option would be to drop an old thread/frame when there was already a newer one (can be checked by comparing frame ids)

pawel-ziecina, 2019-01-07:
I thought that purpose of `on_tick` is some kind of "annoucement" that clock has ticked and further data will be for new frame. And here in addition time is moving back :) 

I also tried to use method of obtaining timestamp for sensor data (as described in #1105) but noticed that sometimes frames are received before `on_tick` callback.

But if server callbacks synchronization is trade-off with server performance - I'd prefer better performance.

One more question - is `world.wait_for_tick` waiting for future ticks or any (also historic ones)?

YashBansod, 2019-01-08:
@pawel-ziecina the images are received using the `sensor.listen` which runs in parallel to `world.on_tick`. Therefore even if you are getting a sensor data with a frame id before the same frame id is printed from world.on_tick, if the id's are same, they are essentially from the same time step.

pawel-ziecina, 2019-01-08:
Sure I understand the cause of this, and how to handle it.

My issue is that this behaviour is not obvious and make client code more complex. And maybe it does not have to be like that.

At least we've documented this behaviour in this issue :)

closed 2019-01-08 by pawel-ziecina