carla-simulator/carla #1109: spawn_npc.py -n 80 not working

MrYouMath, 2019-01-08:
I am trying to run Carla 0.92 following [the documentation](https://carla.readthedocs.io/en/latest/getting_started/).

I downloaded the latest release from the [GitHub page](https://github.com/carla-simulator/carla/releases). I unzipped the CARLA_0.9.2.tar.gz file. Then I opened a terminal inside the folder to run './CarlaUE4.sh' which works. I can fly through the city without any problems. Then I tried to run 'spawn_npc.py -n 80' (also tried './spawn_npc.py -n 80' still inside the folder. But I get the following error:

```
dom@dom:~/carla-dagger$ ./spawn_npc.py -n 80
Traceback (most recent call last):
  File "./spawn_npc.py", line 23, in <module>
    import carla
ModuleNotFoundError: No module named 'carla'
```
I tried to follow 'https://github.com/carla-simulator/carla/issues/121' and 'https://github.com/carla-simulator/imitation-learning/issues/2' but I think these posts are related to an older version.

Then I went download the GitHub page 'https://github.com/carla-simulator/carla'. I took this folder and copied it into the folder which does have the 'CarlaUE4.sh' and the 'spawn_npc.py -n 80' (the release download) and renamed the folder from 'carla-master' to 'carla' and tried 'spawn_npc.py -n 80' but that gave me the output and error

```
destroying 0 actors

done.
Traceback (most recent call last):
  File "./spawn_npc.py", line 123, in <module>
    main()
  File "./spawn_npc.py", line 66, in main
    client = carla.Client(args.host, args.port)
AttributeError: module 'carla' has no attribute 'Client'
```

Then I found out that inside[ `depracated/PythonClient`](https://github.com/carla-simulator/carla/tree/master/Deprecated/PythonClient) a folder named 'carla' was there. I renamed the 'carla' folder back to 'carla-master' and copied the 'carla' folder from the Depracated/PythonClient' into the main folder from the release to subsequently run 'spawn_npc.py -n 80' but that gave me the same error as in the previous step.

Can someone tell me what exactly is going wrong? How should my folder structure look like and what do I need to download into the folder?

tanglrHello, 2019-01-09:
I am on release 0.9.1, but I think there may not be much difference. The carla module are included in the directory 'PythonAPI', in which you can see two .egg files. And in spawn_npc.py, take a look at the lines before 'import carla'. There is a sys.path.append() call, and this is used to indicate the path of the egg files. I think the problem may be your sys.version_info is not consistent with that in the name of the egg files. Maybe you can hardcode the path in sys.path.append() and try again?

MrYouMath, 2019-01-09:
@tanglrHello: Thank you for your help. I tried to change

```
try:
    sys.path.append(glob.glob('**/*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
```
to

```
try:
    sys.path.append(glob.glob('/PythonAPI/carla-0.9.2-py3.5-linux-x86_64.egg'))
except IndexError:
    pass
```

but it still returns me an error `ModuleNotFoundError: No module named 'carla'`

Fenix0817, 2019-01-09:
Hello everybody,

I have the same problem with the module 'carla' but working with the example in the ros-brigde documentation (https://github.com/carla-simulator/ros-bridge).

nsubiron, 2019-01-09:
At the moment we only compile the carla module for Python 2.7 and 3.5, either run it with one of those versions or compile it from source for the Python in your environment.

anshulpaigwar, 2019-01-10:
For the error `ModuleNotFoundError: No module named 'carla'` many people having this issue. The solution is:
```
sudo easy_install <path to .egg file>
```

Also you can modify the python path
```
export PYTHONPATH=$PYTHONPATH:<path/to/carla/>/PythonAPI/<your_egg_file>
``` 

This is not properly documented

Fenix0817, 2019-01-10:
Thank you so much https://github.com/anshulpaigwar. It is working for me.

anshulpaigwar, 2019-01-10:
Please check the pull request #1118. I updated the documentation.

nsubiron, 2019-01-10:
@anshulpaigwar I don't think that's really a fix, if you do an easy_install of the 3.5 module in say Python 3.7, it may work but it may very well crash, in principle they're not compatible. If you have the right version of Python our method of importing Carla should work without error.

anshulpaigwar, 2019-01-10:
@nsubiron I agree it is not a fix for the problem of python 3.5 and 3.7. But in general when we download the Carla binaries `python spawn_npc.py -n 80` doesn't work out of the box even if you have the right version of python (2.7 or 3.5). You need to easy_install the egg files.

nsubiron, 2019-01-10:
Never had that problem, it's usually the other way around, if you install one version of Carla then you are not able to run other versions because the one you have installed is always selected instead and you get weird errors.

Fenix0817, 2019-01-10:
Sorry @nsubiron, but I had this problem. The solution presented by @anshulpaigwar working for me. I also believe that something is wrong with the documentation. 

Have a good day.

closed 2019-03-18 by stale[bot]

oemer90, 2019-04-02:
Do you know, how I can fix this on Windows? Many thanks in advance ;-)

vrajendrareddy, 2019-05-24:
Please suggest how to fix this issue in windows - spawn_npc.py -n 80 - carla not found

autonomobil, 2019-08-19:
I had a similar issue in a conda enviroment. This was my solution:
 
- locate .egg-files in `Carla/PythonAPI/carla/dist/`
- change/install python version in conda enviroment which fits closest to an .egg-file (for me carla-0.9.6-py3.5-linux-x86_64.egg -> 3.5)
- delete all other .egg-files (just for tidiness)
- extract your .egg-file, e.g. carla-0.9.6-py3.5-linux-x86_64
- create `setup.py` inside the folder (carla-0.9.6-py3.5-linux-x86_64) with the following content:

```
from distutils.core import setup
setup(name='carla',
      version='0.9.6', #doesn't matter I guess
      py_modules=['carla'],
      )
```
- then you can install the carla-python-package via pip: 
`pip install -e ~/Carla/PythonAPI/carla/dist/carla-0.9.6-py3.5-linux-x86_64`

That solved it for me. I don't even need the egg-path-appending-stuff, which is happening at the beginning of every python-script, just `import carla.`

pirate-lofy, 2020-03-18:
The solution which worked for me is to leave the .egg files and install the library Carla with pip and it worked without appending the .egg path to the system path

lydiaCS, 2021-02-09:
Egg file of mine is carla-0.9.11-py3.7-win-amd64, so I should use python version 3.7 instead of any other updated versions. and it works

elvira3871, 2021-02-22:
> 我的Egg文件是carla-0.9.11-py3.7-win-amd64，因此我应该使用python版本3.7而不是其他任何更新版本。它有效

> Egg file of mine is carla-0.9.11-py3.7-win-amd64, so I should use python version 3.7 instead of any other updated versions. and it works
I encountered ModuleNotFoundError：there is no module named "carla" when I was building Carla0.9.11 on Windows. According to the tutorial on the official website, I have tested that the python version in the .egg is consistent with the python version in the environment variable. I want to ask you if Learn more about solutions.