carla-simulator/carla #1110: error when i make launch,

Firesuiry, 2019-01-08:
the error is 
`make: *** No rule to make target 'launch'.  Stop.`
How can i fix this error?
thanks a lot..

tanglrHello, 2019-01-09:
Could you give more detail? In which path you make launch? And what you have in that path?

Firesuiry, 2019-01-09:
thank you for your reply.
I make launch at carla file.
the file content is as follow:
[image]
the files is too many because I have depress some assets files wrongly into this file.
and my unreal is installed successfully ,the version is 4.19.2.

tanglrHello, 2019-01-09:
Which version for carla are you using? It seems a little bit different from release0.9.1 I'm using. 
[image]

This error seems to be related with Makefile. My Makefile content:
[image]

And I'm on Linux, so take a look at Util/BuildTools/Linux.mk, you will see the 'lanuch' target. In my enviroment, if I execute "make hello" for example, I'll get the same error as yours.

Firesuiry, 2019-01-09:
i am using the release 0.8.2
and my Makefile is different with yours.
Maybe this will cause the error,i will try the release 0.9.2.
thank you for your help!!

closed 2019-03-17 by stale[bot]

MSiam, 2019-07-04:
I had the same error when I checked out the stable branch but with the master branch the error was gone.

soheilade, 2020-03-02:
Dear All,

I am following the tutorial [here](https://carla.readthedocs.io/en/latest/how_to_build_on_linux/) to build carla. However, running make launch fails on branch stable with this error. 
[image]
on the branch master, I fail with this error: 
[image]
any idea how to deal with this?