carla-simulator/carla #1112: Running carla without display

tanglrHello, 2019-01-09:
I've found several ways to start carla server without display:
(1) https://carla.readthedocs.io/en/latest/carla_headless/
In this documentation, several tools are required to be installed
(2)https://carla.readthedocs.io/en/latest/configuring_the_simulation/#running-off-screen
But in this documentation, only assign an empty value to DISPLAY will do

In my local desktop ubuntu system, the second method seems to work well. But I always get a core dump as described in #1101. 
My situation is: I have several remote headless server machine with nvidia driver installed (but I'm not sure if they're installed with opengl). And I want to start carla on those remote machines. I think I'd better use docker to avoid some common deployment in multiple machines. And if I pull the docker carlasim/carla:0.9.1 from with the instruction in this documentation[https://carla.readthedocs.io/en/latest/carla_docker/#getting-it-running](url), I can run it successfully on my local desktop system, but have signal 11 caught when running on those remote servers. Even the server has a newer nvidia driver than that on my local system.

And to sum up my question:
1. What's the difference behind the two ways to start carla without display?
1. With nvidia driver properly installed on remote server, can I directly use the docker image named  carlasim/carla to start a carla server? Or I should follow the instruction in  https://carla.readthedocs.io/en/latest/carla_headless/ and install those tools into the docker image? Or something else?
3. In what circumstances should I install OpenGL, VirtualGL(VGL), TurboVNC 2.11 mamually? Maybe quite the same as the first one.

fangchuan, 2019-02-20:
hi buddy, have you solved this issue? i occurred the same problems, i also have a remote server equipped with a GPU, but i always got signal 11 or 6 error when i launch the CarlaUE4.sh (version==0.8.2). In addition, my remote server had install the OpenGL 3.0, but it seemed the UE4.18 needs OpenGL4.3 at least.

closed 2019-04-28 by stale[bot]

Derekabc, 2019-06-16:
Hello all, any update on this problem? I am also confused with the same problems proposed by tanglrHello.

qhaas, 2019-07-01:
Same here, @Derekabc , the documentation could use some updates/clarification.  I'm using approach 2 with CARLA 0.9.5's Docker container.  I'm not trying to x-forward the server's display, instead I'm relying on data being dumped to the filesystem via the PythonAPI client.  Seems to work, but some of the messages I'm seeing in the logs make me uneasy (will likely create a separate ticket).

1469cgw, 2020-03-06:
How do I run the Carla training model on a Linux remote server without root access

kimown, 2020-08-11:
> Same here, @Derekabc , the documentation could use some updates/clarification. I'm using approach 2 with CARLA 0.9.5's Docker container. I'm not trying to x-forward the server's display, instead I'm [...]

I have same problem, do you use xorg to start DISPLAY in docker?