carla-simulator/carla #1115: How to understand carla.actor.parent?

JackRen88, 2019-01-10:
Hello,everyone!
When I look through source code in the carla_ros_bridge package,I notice that there is parent or child class and actor.parent.How can I understand this content?

fabianoboril, 2019-01-10:
Hi,
Let's assume you have a sensor attached to a vehicle, for example a camera. In CARLA, both of these are actors. The vehicle is the parent, and the camera is the child, as the camera is attached to the vehicle. Clear?

JackRen88, 2019-01-11:
@fabianoboril 
Thank you.
In parent.py of carla_ros_bridge.there is the comment as follows:
`:param carla_id: unique carla_id of this parent object
            carla_id > 0: carla actor ids (see also carla.Actor)
            carla_id == 0: used by the (root) bridge object
            carla_id == -1: used by the map object`
Can you explain the param carla_id in detail?

closed 2019-03-19 by stale[bot]