carla-simulator/carla #1119: Tcp accept error: Too many open files

nsubiron, 2019-01-10:
As reported a few times, it seems to be a limit on how many sensors can be added to the simulation (even if they're destroyed before creating a new one).

Adding this issue to investigate this problem.

praveen-palanisamy, 2019-01-10:
This bug seems to affect the *number of additions* of new sensors even if only a few (even 1 or 2) are active at any given tick. I suspect that the cleanup/`.destroy` method is not handling the tcp resource releases properly.
CarlaUE4 (Client API) leaves many TCP connections in the `CLOSE_WAIT` state which may be an issue. The attached log file is a sample showing the active TCP connections used by CarlaUE4 which may be useful: [close_wait_zombies.txt](https://github.com/carla-simulator/carla/files/2745741/close_wait_zombies.txt)

praveen-palanisamy, 2019-01-10:
On a related note, a couple of strange observations:
1. Why does CarlaUE4 have to listen to the TCP traffic from all the network interfaces?
```bash
tcp        0      0 0.0.0.0:46983           0.0.0.0:*               LISTEN      138438/CarlaUE4 
tcp        0      0 0.0.0.0:29831           0.0.0.0:*               LISTEN      138486/CarlaUE4 
tcp        0      0 0.0.0.0:46984           0.0.0.0:*               LISTEN      138438/CarlaUE4 
tcp        0      0 0.0.0.0:29832           0.0.0.0:*               LISTEN      138486/CarlaUE4 
tcp        0      0 0.0.0.0:45900           0.0.0.0:*               LISTEN      138536/CarlaUE4 
tcp        0      0 0.0.0.0:45901           0.0.0.0:*               LISTEN      138536/CarlaUE4 
tcp        0      0 0.0.0.0:47447           0.0.0.0:*               LISTEN      137353/CarlaUE4 
tcp        0      0 0.0.0.0:47448           0.0.0.0:*               LISTEN      137353/CarlaUE4 
```

2. Connections with these 2 foreign addresses are necessary/inevitable? 
 - The `52.*` seems to be some AWS node. UE4 crash reporter?
```bash
tcp       32      0 166.166.2.9:44144       52.4.127.66:443         CLOSE_WAIT  138486/CarlaUE4 
tcp       32      0 166.166.2.9:44136       52.4.127.66:443         CLOSE_WAIT  138438/CarlaUE4
tcp       32      0 166.166.2.9:44146       52.4.127.66:443         CLOSE_WAIT  138536/CarlaUE4
tcp       32      0 166.166.2.9:49762       100.25.57.141:443       CLOSE_WAIT  137353/CarlaUE4
```

nsubiron, 2019-01-10:
Uhm I'll take a look into that too.

The foreign addresses are almost for sure the crash reporter, I believe we can remove the crash reporter, we don't use it.

nsubiron, 2019-01-11:
I think I've identified the problem. It only happens with the collision sensor, doesn't seem to happen with the cameras.

When the client closes the socket, the socket remains in CLOSE_WAIT state until the server also closes it. In our library, the server only checks the state of the socket when it tries to send some data, thus the next time the sensor produces some data, it will notice that the socket was closed by the client and it will close it too. Since cameras are sending data each frame asynchronously the socket is always closed (there is always some data that is going to be sent after destroy). With the collision sensor however, the moment we destroy the sensor, no more data is going to be sent, thus the server session won't notice that the socket was closed and the socket will remain in CLOSE_WAIT state.

The "destroy" process looks something like this

1. Client closes streaming socket
2. Client request sensor destruction via RPC
3. Server destroys sensor
4. If there is more data
    - Failure to send because client closed socket
    - Server closes socket too

We need to close the socket in the server-side after step 3.

Thanks for the CLOSE_WAIT hint!

closed 2019-01-29 by nsubiron