carla-simulator/carla #1120: Coordinate system relative to the car

atanas1054, 2019-01-10:
Hi,
I want to change the coordinate system such that the (0,0,0) point is the location of the sensor attached to the car. So X direction will be forward (direction of travel), Y direction will be sideways (left|right) and Z will be height. And then I want to get the location of all pedestrians relative to that point. How can I do that?

Unlingius, 2019-01-10:
As you have camera transform from Carla, you may do it in your program. It looks something like:
xyz_relative = R * (xyz_pedestrian - xyz_car), where R is orientation of camera.   Or may be transpose(R), try

atanas1054, 2019-01-11:
Thank you for your reply. Can you tell me how to get the orientation of the camera R? What is the size of R? Also what is the operation *?

Unlingius, 2019-01-11:
see manual_control.py example in pythonAPI, line 165 - getting pose of car,  and line 632 - image.transform is pose of camera. Note, that Carla use left coordinate system, so first you have to change it to right: x,y,z -> x,-y,z, and some transformations of angles. Then you should form 3x3 matrix R from that. For this purpose I use transforms3d package for python. It is something like: R = transforms3d.euler.euler2mat(roll, pitch, yaw).T
xyz_pedestrian and car also should be in the same system. * is multiply matrix and vector.

atanas1054, 2019-01-15:
Ok, I tried ped_loc_relative = numpy.dot(R, ped_loc)

Also, I have fixed the relative position of the pedestrian to the car in Z direction by fixing the roll and the pitch to 0 (car doesn't move much around those axis). However, the X and Y axis are interchanging /acting weird when the car is turning.

Unlingius, 2019-01-15:
vehicle_xyz and xyz_car are the same in your code?

atanas1054, 2019-01-15:
I am doing exactly the same. The only difference is that in carla the angles are given in degrees and not radians.

Unlingius, 2019-01-15:
have you turned it to radians?

atanas1054, 2019-01-16:
That was it! I was using degrees instead of radians. It is working now, thank you again.

closed 2019-01-16 by atanas1054

1453042287, 2019-03-23:
@Unlingius thanks a lot！