carla-simulator/carla #1122: tf transform between /ego_vehicle and /ego_vehicle/lidar/front/ is not fixed

JackRen88, 2019-01-11:
Hello,Carlateam!
When I run the carla_ros_bridge in the version 0.9.2 of Carla,I look at the tf transform between /ego_vehicle and /ego_vehicle/lidar/front/,and find that when the ego vehicle keeps still,the tf transform is fixed,but when the  ego vehicle moves ,the tf transform is not fixed and is changing real time.
So lidar is attached to the ego vehicle,shouldn't the tf transform between keep constant?

fabianoboril, 2019-01-11:
Hi,
are you really looking at the transform between the car and the sensor, or only at the transform of each of them? The transform of the sensor is w.r.t the world frame!

JackRen88, 2019-01-11:
@fabianoboril 
When I run the client.launch in the carla_ros_bridge and manual_control.py, I run the following node:
`rosrun tf tf_echo /ego_vehicle /ego_vehicle/lidar/front/` to print the tf transform between /ego_vehicle and /ego_vehicle/lidar/front,finding the above issues.

closed 2019-03-19 by stale[bot]