carla-simulator/carla #1125: Error when importing tensorflow before carla

pawel-ziecina, 2019-01-11:
Hi,
I noticed serialization (?) error when importing tensorflow before carla.

Code to reproduce:
```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import tensorflow as tf
import carla
print(carla.Location())
```

Output:
```
Traceback (most recent call last):
  File "./foo.py", line 7, in <module>
    print(carla.Location())
RuntimeError: std::bad_cast
```

When move carla import before tensorflow it works ok.

OS: Ubuntu 18.04
carla: 0.9.2 official build
tensorflow: tensorflow-gpu==1.12.0

coreper, 2019-01-11:
I have a similar problem with Pytorch 1.0. I have to reinstall pytorch 0.4 to solve this problem.

kk2491, 2019-02-12:
I am facing the same issue when I try to do the below lines of code 

```
>>> import carla
>>> 
>>> client = carla.Client('localhost', 2000)
>>> client.set_timeout(10.0) 
>>> 
>>> world = client.get_world()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: std::bad_cast
>>> 
>>> world = client.get_world()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: std::bad_cast
>>> 
```

Thank you

pawel-ziecina, 2019-03-15:
Still observing `std::bad_cast` error with below snippet on 0.9.4 official build with tensorflow-gpu==1.12.0

```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import tensorflow as tf
import carla
print(carla.Location())
```
@kk2491 could not reproduce your issue:

```python
import carla
client = carla.Client('localhost', 2000)
client.set_timeout(10.0) 
world = client.get_world()
```

kk2491, 2019-03-15:
I was able to fix that issue, but I don't remember what exactly the problem was.

Can you try this.. please check if there is any previous version of Carla installed in your system? I remember like I had to delete the older version which I built from source.

Let me know if it works.

Thank you,
KK

pawel-ziecina, 2019-03-15:
I haven't got any reference to other carla release in my system. Also above snippet was checked on other workstations.

@kk2491 Just check my snippet and let me know if it fails or not. I think it is bug not related with environment.

If needed my venv preparation:

```shell
cd <carla_release_dir>
python3 -m venv .venv
source .venv/bin/activate
pip install -r tensorflow-gpu==1.12.0
PYTHONPATH=$PWD/PythonAPI/carla-0.9.4-py3.5-linux-x86_64.egg python
```

jkewang, 2019-03-25:
Same error! when I import tensorflow 
Before Carla:
runtimeError: std::bad_cast
After Carla:
Non-Ok-status: RegisterAlreadyLocked()……

fedor-chervinskii, 2019-04-04:
Same here with PyTorch:

This is working:
```import carla
import torch

client = carla.Client('localhost', 2000)
client.set_timeout(10.0)
world = client.get_world()
print("success!")```

But this fails with timeout error:

```import torch
import carla

client = carla.Client('localhost', 2000)
client.set_timeout(10.0)
world = client.get_world()
print("success!")```

xmyqsh, 2019-06-08:
Same here:

```
import networkx
import carla
print carla.Location()
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: std::bad_cast

```
import shapely.geometry
import carla
print carla.Location()
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: std::bad_cast

closed 2019-08-14 by stale[bot]