carla-simulator/carla #1127: New sensor: obstacle detector

nsubiron, 2019-01-11:
Implement a new sensor that allows checking for obstacles ahead, probably a single ray/sphere cast.

ignacioalvmar, 2019-02-21:
I think the current implementation is not working properly.

I have created a client classs using this obstacle that I want to use for "line of sight":

```
class LineOfSightSensor(object):
    def __init__(self, parent_actor, hud):
        self.sensor = None
        self._history = []
        self._parent = parent_actor
        self._hud = hud
        self._event_count = 0
        self.sensor_transform = carla.Transform(carla.Location(x=1.6, z=1.7), carla.Rotation(yaw=0)) # Put this sensor on the windshield of the car.
        world = self._parent.get_world()
        bp = world.get_blueprint_library().find('sensor.other.obstacle')
        bp.set_attribute('distance', '200')
        bp.set_attribute('hit_radius', '12')
        bp.set_attribute('only_dynamics', 'true')
        #bp.set_attribute('debug_linetrace', 'true')
        bp.set_attribute('sensor_tick', '1')
        self.sensor = world.spawn_actor(bp, self.sensor_transform, attach_to=self._parent)
        weak_self = weakref.ref(self)
        self.sensor.listen(lambda event: LineOfSightSensor._on_LOS(weak_self, event))

    @staticmethod
    def _on_LOS(weak_self, event):
        self = weak_self()
        if not self:
            return
        print (str(event.other_actor))
        if event.other_actor.type_id.startswith('vehicle.'):
            print ("Event %s, in line of sight with %s at distance %u" % (self._event_count, event.other_actor.type_id, event.distance))
            self._event_count += 1
```
I am assuming that specifying the distance as 200 is the "length" of the raycast and hit_radius is the "thickness" of the line.

The results that I am getting are confusing:

Here some output logs:

```
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=0, type=static.vegetation)
Actor(id=55, type=vehicle.nissan.patrol)
Event 0, in line of sight with vehicle.nissan.patrol at distance 1200
Actor(id=55, type=vehicle.nissan.patrol)
Event 1, in line of sight with vehicle.nissan.patrol at distance 1200
Actor(id=55, type=vehicle.nissan.patrol)
Event 2, in line of sight with vehicle.nissan.patrol at distance 1200
Actor(id=55, type=vehicle.nissan.patrol)
Event 3, in line of sight with vehicle.nissan.patrol at distance 1200
Actor(id=55, type=vehicle.nissan.patrol)
Event 4, in line of sight with vehicle.nissan.patrol at distance 1200_
```

with the 'only_dynamics' flag aren't I suppose to see only vehicles? why Do I have static.vegetation?

If I leave the 'debug_linetrace' flag uncommented the rendering on the client breaks. this is the output on the carla server:
```
[2019.02.21-19.27.40:351][531]LogCarla: Spawning actor 'sensor.other.obstacle'
[2019.02.21-19.27.40:610][533]LogCarla: Spawning actor 'sensor.camera.rgb'
[2019.02.21-19.27.40:610][533]g.TimeoutForBlockOnRenderFence = "300000"
[2019.02.21-19.27.40:790][535]LogRenderer: Reallocating scene render targets to support 1280x720 Format 10 NumSamples 1 (Frame:38535).
[2019.02.21-19.28.12:798][540]LogTaskGraph: Wa
[2019.02.21-19.28.12:798][540]LogTaskGraph: Warning: Task graph took 277.21ms for RT to recieve broadcast and do processing.
[2019.02.21-19.28.12:798][540]LogTaskGraph: Warning: Task graph took 277.27ms to wait for named thread broadcast.
[2019.02.21-19.28.12:798][540]LogTaskGraph: Warning: Task graph took 277.48ms to broadcast.
rning: Task graph took 277.05ms for RT to recieve broadcast.
```

I think there is a render method missing so that we can activate the visualization of the debug linetrace.

Derekabc, 2019-07-02:
have you solved the problem, I find the same problem

roser-br, 2020-01-14:
Hi @Derekabc and @ignacioalvmar! 
Were you able to solve this problem? I am able to identify different obstacles on the way, but I keep getting the same distance, it never changes. 
Or do you have any tip on how to obtain the distance from the car to the upcoming obstacle? 
Thank you in advance!

Derekabc, 2020-01-14:
Can you check this file: [here](https://github.com/Derekabc/RL/blob/master/lane_changing-v2.py). Here I change the lane when detect the obstacle vehicle ahead of us. Hope it can help you!
`_is_vehicle_hazard(world, max_distance):`
This function is used to detect obstacles and the main function provides the logic. 
Thanks!

omkukunuru, 2020-04-10:
@roser-br I am also facing the same problem. I keep getting the same distances.
I see that the link provided by @Derekabc should have a fix. But, getting a 404 error while accessing the page. Can you direct me to correct page or can you say how it has fixed your problem?