carla-simulator/carla #1133: anti-aliasing

Unlingius, 2019-01-14:
Hi!
I observe, that anti-aliasing function is turned on in main window, but not in client.
[image]
is there any to turn on it? Thanks.

manishthani, 2019-01-14:
Hi @Unlingius
 
Right now this functionality is not available due to some issues related with Unreal engine built on Linux that makes the server send the frame to the client **without** some post processing effects, where one of them is anti-aliasing. 
As an alternative, it could be possible to use [**FXAA**](https://developer.download.nvidia.com/assets/gamedev/files/sdk/11/FXAA_WhitePaper.pdf) post processing effect on the received frame from the server which is **already** implemented by Nvidia in their [Github repository](https://github.com/NVIDIAGameWorks).

Hope this helps!
Manish

Unlingius, 2019-01-15:
@manishthani  Thank you for reply. Can you recommend any easy-to-use tutorial how to try it on given frame? I even don't need realtime speed.

Unlingius, 2019-01-15:
I tried SMAA morphological aintialising from https://github.com/iRi-E/smaa-cpp and got result:
[image]
It works good with "stairs", but not with gaps. It seems, any postprocessing methods are not enough in this case. I found, that maybe useful follows methods: tesselation shaders, conservative rasterization, SSAA, MSAA. They are not postprocessing. Is that possible to activate it in unreal settings?

Unlingius, 2019-01-18:
to deal with gaps for now I use supersampling, i set_attribute('image_size') of cameras bigger, then i need, and then use opencv resize function with interpolation. It is expensive and not realtime, but work.

closed 2019-03-26 by stale[bot]