carla-simulator/carla #1134: Missing rain in horizontal bar at eye level

martijn-vinotion, 2019-01-14:
Hi all,

When enabling precipitation, I notice that the rain is not rendered at the horizontal bar at eye level of the player. This can be seen in the screenshot below.
[image]
Can anyone help me solving this problem?

I am using Carla 0.9.0.

Thanks in advance!

TheNihilisticRobot, 2019-01-15:
Hi @martijn-vinotion 

This is probably because you've asked the engine for more rain particles than your current computer can process.

When the engine runs out of Cpu power to spawn rain particles it stops spawning until enough particles have despawned then starts again creating this "gaps" between spawn periods.

For now I'll suggest toning down the rain parameter until you don't see this effect anymore. I can't give you an exact number because there are a lot of factors to consider here, I think trial and error will be a better approach.

martijn-vinotion, 2019-01-28:
That solved the problem. Thank you very much!

closed 2019-01-28 by martijn-vinotion