carla-simulator/carla #1135: Can you provide the visualization of spawn points in Map03

Johnnyswong, 2019-01-15:
Hi, now i'm training my model with Carla in Map03, and I want to start in a fixed road structure, eg. straight road. But this function requires the config for spawn points in Map03. 

Map01 and Map02 is provided in the documentation but Map03 is not.

Sakrust, 2019-01-18:
You can use get_spawn_points() to retrieve a list of all spawn points and then draw them on your map with the debug helper functions.
Or open your Unreal Project in editor and see where the SpawnPoints are placed.

kartik144, 2019-03-14:
> You can use get_spawn_points() to retrieve a list of all spawn points and then draw them on your map with the debug helper functions.
> Or open your Unreal Project in editor and see where the SpawnPoints are placed.

Hi @Sakrust 
I am able to use get_spawn_points() to get all the spawn points on the map. But I am not able to figure out how to draw them on the map. I tried using the `draw_point()` function in `carla.DebugHelper`, but I don't think I was using it the right way, and got the following error:
```
carla.DebugHelper.draw_point(p.location)
Boost.Python.ArgumentError: Python argument types in
    DebugHelper.draw_point(Location)
did not match C++ signature:
    draw_point(carla::client::DebugHelper {lvalue}, carla::geom::Location location, float size=0.10000000149011612, carla::sensor::data::Color color=<carla.libcarla.Color object at 0x7f6cb82853a0>, float life_time=-1.0, bool persistent_lines=True)
```
Could you please help me out here?

Sakrust, 2019-03-14:
I personally don't use the draw_point function, but prefer to use draw arrow or draw string, because of their better visibility.
```
for waypoint in map.get_spawn_points():
world.debug.draw_string(waypoint.transform.location, 'o', draw_shadow=False,
											 color=carla.Color(r=255, g=255, b=255), life_time=0.2,
											 persistent_lines=True)
```
This code snippet should fit your needs.

kartik144, 2019-03-14:
> I personally don't use the draw_point function, but prefer to use draw arrow or draw string, because of their better visibility.
>
> ```
> for waypoint in map.get_spawn_points(): [...]

@Sakrust thanks a lot!

closed 2019-05-20 by stale[bot]

Kin-Zhang, 2022-01-16:
> Hi @Sakrust I am able to use get_spawn_points() to get all the spawn points on the map. But I am not able to figure out how to draw them on the map. I tried using the `draw_point()` function in [...]

I also met the problem, and I notice is because of color should be also write as `carla.color(255,0,0)`
For example:
```python
self._world.debug.draw_point(location, size=3,  color=carla.Color(0,0,255),life_time=100000)
```

TauJiang, 2023-01-07:
> Hi @Sakrust I am able to use get_spawn_points() to get all the spawn points on the map. But I am not able to figure out how to draw them on the map. I tried using the `draw_point()` function in [...]

hi this is because we need to initialize the object so that it has 'self' when it is passed to the arguments.

Liza23, 2023-08-03:
Hi there,
I am facing the same issue as stated above. I am trying to use debugHelper as follows:
```
spawn_points = world.get_map().get_spawn_points()
sample = spawn_points[1].location
carla.DebugHelper.draw_point(carla.Location(sample), size=0.05, life_time=7200, color=carla.Color(238, 18, 137, 0))

```

However, I am facing exactly the same issue 
```
carla.DebugHelper.draw_point(p.location)
Boost.Python.ArgumentError: Python argument types in
    DebugHelper.draw_point(Location)
did not match C++ signature:
    draw_point(carla::client::DebugHelper {lvalue}, carla::geom::Location location, float size=0.10000000149011612, carla::sensor::data::Color color=<carla.libcarla.Color object at 0x7f6cb82853a0>, float life_time=-1.0, bool persistent_lines=True)
```

Let me know if you find anything useful.