carla-simulator/carla #1137: scenario runner ImportError: No module named carla

geekhut, 2019-01-16:
Hi all, new to carla here. I am looking through this tutorial on scenario runner,

https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_started.md

and I ran into the no module named carla again, I have already looked at the relevant issues raised by others but still, I can't seem to solve this problem. Help me out please!

fabianoboril, 2019-01-16:
Hi,
did you set your PYTHONPATH as mentioned in the documentation?
`export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla-<VERSION>.egg`

geekhut, 2019-01-16:
> Hi,
> did you set your PYTHONPATH as mentioned in the documentation?
> `export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla-<VERSION>.egg`

Hi, I have settled the issue, by setting the right path and using python3. Python2 leads to "importError: No module named carla".

fabianoboril, 2019-01-17:
It all depends on which egg you use. If you want to use Python2, you need to use the egg for Python2.

wendersonj, 2019-02-07:
This information could be in the Getting Start from the lastest documentation, causa I didn't found it and was really important ! Anyway, thanks @geekhut !

wendersonj, 2019-02-07:
This information could be in the Getting Start from the lastest documentation, because I didn't found it and was really important ! Anyway, thanks @fabianoboril !

closed 2019-04-15 by stale[bot]

vishal-prgmr, 2019-05-09:
@fabianoboril , where do i find the egg files. I built Carla 0.9.5 from source for Windows.
Running make PythonAPI returns the error "Make sure "py" is installed but i am calling this command from inside a environment which has python3.7.3"

Sibi-ROS-coder, 2021-03-27:
> > Hi,
> > did you set your PYTHONPATH as mentioned in the documentation?
> > `export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla-<VERSION>.egg`
> 
> Hi, I have settled the issue, by setting the right path and using python3. Python2 leads to "importError: No module named carla".

well i did the same thing the import Carla problem pops can you be specific what to do for pythin2 as well I am new to carla