carla-simulator/carla #1140: Steps to fix "AttributeError: 'module' object has no attribute 'Client'"

kk2491, 2019-01-17:
Hi,

This issue has been mentioned already. However I would like to provide the sequence of steps I used to fix the issue.

I have downloaded the latest pre-compiled version of Carla from [link](https://drive.google.com/file/d/1Wt2cxXCtWI3cSI4rt3_HjGnVfkK8Z9bl/view).

Started the simulator by running ```./CarlaUE4.sh``` in terminal.

In new terminal started ```python spawn_npc.py -n 80``` to add vehicles to the simulator.
This would result in below error.
**Error1**
```
user@system:~/Carla_Sim$ python spawn_npc.py -n 80

destroying 0 actors

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 123, in <module>
    main()
  File "spawn_npc.py", line 66, in main
    client = carla.Client(args.host, args.port)
AttributeError: 'module' object has no attribute 'Client'
```
This is probably due to the reason that I had built the older carla version previously 

Try running the below to check if it pointing to the right version.
```
user@system:~/Carla_Sim$  python -c 'import carla; print carla'
<module 'carla' from '/home/kishor/carla/PythonClient/carla/__init__.pyc'>
```
It is pointing to the wrong Carla version, you can delete or rename the older carla version to fix the **Error1**.

Running the spawn script again would result in below error.

**Error2**
```
user@system:~/Carla_Sim$ python3 spawn_npc.py -n 80
Traceback (most recent call last):
  File "spawn_npc.py", line 23, in <module>
    import carla
  File "PythonAPI/carla-0.9.2-py3.5-linux-x86_64.egg/carla/__init__.py", line 7, in <module>
  File "PythonAPI/carla-0.9.2-py3.5-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "PythonAPI/carla-0.9.2-py3.5-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
  File "/usr/lib/python3.5/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: libpng16.so.16: cannot open shared object file: No such file or directory
```
You can install it by running ```sudo apt-get install libpng16-16```.

Finally you can run ```user@system:~/Carla_Sim$ python3 spawn_npc.py -n 80``` again to get the vehicles on simulator.

Thank you,
KK

gnykumar, 2019-02-19:
Thanks. It resolved my first error

closed 2019-04-27 by stale[bot]

iamycee, 2019-08-30:
> It is pointing to the wrong Carla version

Which is the correct version?