carla-simulator/carla #1144: Generate OpenDrive XML format from Carla map

BoChang11, 2019-01-17:
Hi Guys,

I have seen in Carla documents that maps used in Carla come accompanied by OpenDrive files. Kindly help in understanding the below queries.

1. Kindly let me know the location of the OpenDrive XML map files.
2. If XML is not present, is there a way I could generate the XML map file from the Carla layout.

BR

Sakrust, 2019-01-18:
1. Content/Carla/Maps/OpenDrive

2. You can use the functions provided: 
to_opendrive()
save_to_disk(path=self.name)

or use RoadRunner to create a Map and export the map as OpenDrive.

@https://carla.readthedocs.io/en/latest/python_api/

BoChang11, 2019-01-24:
@Sakrust Thanks for the answer. I was trying to integrate Apollo and Carla.
And hoping that the map from Carla is in same format as in Apollo Auto, so that I can use the map in Apollo simulation. But the Carla uses default open drive specification, where as Apollo uses custom open drive format.

BR

closed 2019-01-24 by marcgpuig