carla-simulator/carla #1147: Newest version of exe suddenly crash on Win 10

didibaba, 2019-01-17:
Hi, 

I have successfully built Carla on win 10. However, after double click exe,  only the welcoming image shows up, and then the process crashes.  

I am wondering if there is something wrong with the compiling.

Thx.

[image]

didibaba, 2019-01-17:
[image]

closed 2019-01-18 by didibaba

gaurish123, 2019-01-18:
Hi, I am getting this issue while trying to build carla on my windows,

https://github.com/carla-simulator/carla/issues/1142

didibaba, 2019-01-20:
For the newest version in master branch,  I made two modifications.

1. /carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Carla.h

> #include "Util/NonCopyable.h"
> #include "EngineMinimal.h"
> #include "Logging/LogMacros.h"
> #include "Modules/ModuleInterface.h"

2. /AppData/Roaming/Unreal Engine/UnrealBuildTool/BuildConfiguration.xml

> <?xml version="1.0" encoding="utf-8" ?>
> <Configuration xmlns="https://www.unrealengine.com/BuildConfiguration">
>     <VCProjectFileGenerator>
>         <Version>VisualStudio2017</Version>
>     </VCProjectFileGenerator>
> 
>     <WindowsPlatform>
>         <Compiler>VisualStudio2017</Compiler>
>     </WindowsPlatform>
> </Configuration>

PS. I also update my vs2017 to the newest version.

Hope this will help.

didibaba, 2019-01-20:
Hi @gaurish123 ,
One more thing, `make PythonAPI` still failed in this version as I mentioned in #976.  I think  these are some problems with libboost. Part of the error as below.

> C:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt\stdlib.h(1015): note: see declaration of 'wcstombs'
> dependencies/include\boost/python/class.hpp(497): error C2672: 'boost::python::detail::get_signature': no matching overloaded function found
> dependencies/include\boost/python/class.hpp(233): note: see reference to function template instantiation 'void boost::python::class_<carla::client::ActorAttribute,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified>::def_impl<carla::client::ActorAttribute,F,boost::python::detail::def_helper<const char *,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified>>(T *,const char *,Fn,const Helper &,...)' being compiled
>         with
>         [
>             F=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const,
>             T=carla::client::ActorAttribute,
>             Fn=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const,
>             Helper=boost::python::detail::def_helper<const char *,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified>
>         ]
> dependencies/include\boost/python/class.hpp(231): note: see reference to function template instantiation 'void boost::python::class_<carla::client::ActorAttribute,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified>::def_impl<carla::client::ActorAttribute,F,boost::python::detail::def_helper<const char *,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified>>(T *,const char *,Fn,const Helper &,...)' being compiled
>         with
>         [
>             F=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const,
>             T=carla::client::ActorAttribute,
>             Fn=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const,
>             Helper=boost::python::detail::def_helper<const char *,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified>
>         ]
> d:\intelligent_vehicle\carlar-sim\carla\pythonapi\source\libcarla\Blueprint.cpp(115): note: see reference to function template instantiation 'boost::python::class_<carla::client::ActorAttribute,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified> &boost::python::class_<carla::client::ActorAttribute,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified>::def<bool(__cdecl carla::client::ActorAttribute::* )(const T &) const>(const char *,F)' being compiled
>         with
>         [
>             T=bool,
>             F=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const
>         ]
> d:\intelligent_vehicle\carlar-sim\carla\pythonapi\source\libcarla\Blueprint.cpp(132): note: see reference to function template instantiation 'boost::python::class_<carla::client::ActorAttribute,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified> &boost::python::class_<carla::client::ActorAttribute,boost::python::detail::not_specified,boost::python::detail::not_specified,boost::python::detail::not_specified>::def<bool(__cdecl carla::client::ActorAttribute::* )(const T &) const>(const char *,F)' being compiled
>         with
>         [
>             T=bool,
>             F=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const
>         ]
> dependencies/include\boost/python/class.hpp(497): error C2782: 'boost::mpl::vector17<RT,most_derived<Target,ClassT>::type&,T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14> boost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14) volatile const,Target *)': template parameter 'ClassT' is ambiguous
> dependencies/include\boost/python/signature.hpp(236): note: see declaration of 'boost::python::detail::get_signature'
> dependencies/include\boost/python/class.hpp(497): note: could be 'carla::client::ActorAttributeValueAccess'
> dependencies/include\boost/python/class.hpp(497): note: or       'carla::client::ActorAttribute'
> dependencies/include\boost/python/class.hpp(497): error C2784: 'boost::mpl::vector17<RT,most_derived<Target,ClassT>::type&,T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14> boost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14) volatile const,Target *)': could not deduce template argument for 'RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14) volatile const' from 'Fn'
>         with
>         [
>             Fn=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const
>         ]
> dependencies/include\boost/python/signature.hpp(236): note: see declaration of 'boost::python::detail::get_signature'
> dependencies/include\boost/python/class.hpp(497): error C2780: 'boost::mpl::vector17<RT,ClassT&,T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14> boost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14) volatile const)':  [... cut ...]
> dependencies/include\boost/python/class.hpp(497): error C2784: 'boost::mpl::vector17<RT,most_derived<Target,ClassT>::type&,T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14> boost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14) const,Target *)': could not deduce template argument for 'RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7,T8,T9,T10,T11,T12,T13,T14) const' from 'Fn'
>         with
>         [
>             Fn=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const
>         ]

f5maki, 2019-01-23:
Make PythonAPI - to compile on Windows 10.
#976 from iFusionzz has the steps you need to update your <CARLA_INSTALL_DIR>PythonAPI\setup.py and compile zlib and png.

In <CARLA_INSTALL_DIR>\PythonAPI\source\libcarla\Blueprint.cpp comment out
lines 115 to 126 the templated equal and not equal operators. The MSVC compiler does not like these  overloaded operators.

Other things to watch for:
Use Python 3.
Make sure your setuptools are up to date - pip install setuptools (to get the correct MSBuild)
Ensure you match the type of linking for your boost/rpc libraries. I normally override the linking type in <CARLA_INSTALL_DIR>PythonAPI\setup.py (ie add /MD to extra_compile_args)
Ensure your PYTHONPATH environment variable is pointing at your chosen Python installation
Ensure %PYTHONPATH% is in PATH.
Ensure your PATH does not include other python installations. For example I had a Sumo installation with a Python.exe and scripts.