carla-simulator/carla #1148: different image.transforms with different cars

Unlingius, 2019-01-17:
Hi! I have question about image.transform. I use modified manual_control.py example. 
I set x,y,z roll, pitch, yaw of attached camera to [2.0, 0.0, 1.0, 0.0, 0.0, 0.0], relative to car.
My car is on the flat horizontal road, and i look, what image.transform is received. I expect, that image.transform.z will be 1.0 relative to road (my car is not moving), but it's false, image.transform.z is 1.143. Moreover, then i push backspace to change the car. After it, when car has dropped on the road, i wait for receiving image.transform again. And these transforms are significantly different. Why? It looks like a bug.

Unlingius, 2019-01-17:
i understand, that there are physics, suspension.... But difference of z 0.14 is too huge for that effects i think.

Unlingius, 2019-01-17:
Oh, now i select car "Lincoln Mkz2017".  I observe, that when my car hardly brakes, its nose turn to up, not down.

germanros1987, 2020-03-19:
@marcgpuig do we know if this issue is solved?

closed 2020-06-26 by germanros1987