carla-simulator/carla #1151: Can't get_actors() before get_blueprint_library()

tanglrHello, 2019-01-19:
Here is the code:

import carla
client = carla.Client("127.0.0.1", 2000)
client.set_timeout(2.0)
world = client.get_world()
blueprints = world.get_blueprint_library()   //***
actors = world.get_actors()
print ("actor num:" + str(len(actors)))

If I remove the blueprint line, the actors will be empty. Is it a designed feature? Just report the case.

nsubiron, 2019-01-23:
It seems a matter of timing, can you check if it works replacing the `get_blueprint_library()` by a `time.sleep(1)`?

Internally, when you request the world you subscribe to the world observer sensor that sends the list of actors and their transforms. If you are too fast you may not have received any update yet thus it will appear there are no actors.

tanglrHello, 2019-01-24:
Yes, it works with time.sleep(1) before GetActors. 
Using world.wait_for_tick() works, too.

nsubiron, 2019-01-24:
> Using world.wait_for_tick() works, too.

Good idea :smile:

commit db434b0c (2019-02-15): Fix #1151, wait for tick after subscribing to world observer
commit 6abcb692 (2019-02-22): Fix #1151, wait for tick after subscribing to world observer
PR #1257 (2019-02-23): Improve time-out error messages and fix 1151
#### Description

- Improved time-out error messages, now instead of an rpc error you get the following message
```
RuntimeError: time-out of 2000ms while waiting for the simulator, make sure
the simulator is ready and connected to 127.0.0.1:2000
```
- Print same error message if we meet the time-out in `world.wait_for_tick` (RecurrentSharedFuture), previously the message was very cryptic.
- Wait for tick after subscribing to the world observer to have the world ready after `client.get_world()`, solves issues of having an empty list when calling `wo [... cut]
closed 2019-02-23 by nsubiron