carla-simulator/carla #1155: Cannot launch on win10 with 0.9.2

HeYDwane3, 2019-01-21:
I am using CARLA 0.9.2 on win10

I have read the post that some function is only working on Linux, so I am wondering if it is fixed right now.

when building the 0.9.2, I got the following errors

 "C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\CarlaUE4.sln" (default target) (1) ->
       "C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj" (default target) (2) ->
       (Build target) ->
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1689): error C2039: 'GetObjectW'
       : is not a member of 'TScriptInterface<IBlendableInterface>' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermedi
       ate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1709): error C2039: 'GetObjectW'
       : is not a member of 'TScriptInterface<IBlendableInterface>' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermedi
       ate\ProjectFiles\CarlaUE4.vcxproj]
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(59): error C2039: '_InterlockedCompa
       reExchange': is not a member of 'FWindowsPlatformAtomics' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate
       \ProjectFiles\CarlaUE4.vcxproj]
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(59): error C2665: '_InterlockedCompa
       reExchange': none of the 4 overloads could convert all the argument types [C:\Users\heyd0\Desktop\carla\carla\Unreal\Carla
       UE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(78): error C2039: '__faststorefence'
       : is not a member of 'FWindowsPlatformMisc' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\
       CarlaUE4.vcxproj]
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHICommandList.h(282): error C2039: '_InterlockedIn
       crement': is not a member of 'FWindowsPlatformAtomics' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHICommandList.h(282): error C2665: '_InterlockedIn
       crement': none of the 4 overloads could convert all the argument types [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4
       \Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(37): error C2660:
        'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Project
       Files\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(45): error C2660:
        'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Project
       Files\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(128): error C2660
       : 'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Projec
       tFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(137): error C2660
       : 'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Projec
       tFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(37): error C2660:
        'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Project
       Files\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(45): error C2660:
        'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Project
       Files\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(128): error C2660
       : 'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Projec
       tFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(137): error C2660
       : 'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Projec
       tFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(103): error C2039: '__fast
       storefence': is not a member of 'FWindowsPlatformMisc' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(154): error C2039: '__fast
       storefence': is not a member of 'FWindowsPlatformMisc' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(159): error C2039: '__fast
       storefence': is not a member of 'FWindowsPlatformMisc' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(165): error C2039: '__fast
       storefence': is not a member of 'FWindowsPlatformMisc' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1689): error C2039: 'GetObjectW'
       : is not a member of 'TScriptInterface<IBlendableInterface>' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermedi
       ate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1709): error C2039: 'GetObjectW'
       : is not a member of 'TScriptInterface<IBlendableInterface>' [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermedi
       ate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(37): error C2660:
        'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Project
       Files\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(45): error C2660:
        'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Project
       Files\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(128): error C2660
       : 'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Projec
       tFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(137): error C2660
       : 'GetObjectW': function does not take 0 arguments [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\Projec
       tFiles\CarlaUE4.vcxproj]
         UnrealBuildTool : error : UBT ERROR: Failed to produce item: C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Plugins\
       Carla\Binaries\Win64\UE4-Carla.lib [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.
       vcxproj]
         C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targets(44,5):
        error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.19\Engine\Build\BatchFiles\Build.bat" CarlaUE4 Win64 Develo
       pment "C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild" exited with code 5.
       Please verify that you have sufficient rights to run this command. [C:\Users\heyd0\Desktop\carla\carla\Unreal\CarlaUE4\Int
       ermediate\ProjectFiles\CarlaUE4.vcxproj]

    0 Warning(s)
    27 Error(s)

marcgpuig, 2019-01-23:
Hi @HeYDwane3 
> Please verify that you have sufficient rights to run this command.

Try to run it as admin

f5maki, 2019-01-23:
Workaround : edit <CARLA_INSTALL_DIR>\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Carla.h add #include EngineMinimal.h at line 8.

Workaround to fix make PythonAPI on Windows 10
Ensure you follow the instructions from IFusionz in #976. To get the right setup.py referencing zLib and pnglib.

Comment out lines 115 to 126 the templated not equals and equals overloaded operators in <CARLA_INSTALL_DIR>\PythonAPI\source\libcarla\Blueprint.cpp. 

Things to watch for
Make sure setuptools are installed - pip install setuptools. (To get the right version of MSBuild.exe)
Ensure you have consistent linkage between boost/rpc and carla_client. Add /MD to the extra_compile_args in setup.py if you can't be arsed to recompile boost.
Ensure your environment variable PYTHONPATH points to the correct Python 3 installation.
Ensure %PYTHONPATH% is on your PATH and no other hidden Python installations are referenced on PATH.

PR #1241 (2019-02-27): Basic build system for Windows
#### Description
A basic build system for Windows.

**This PR provides**
- Support for Python3 (`make PythonAPI`)
- Automated download and build of `zlib` and `libpng` needed in Win
- Now Boost is downloaded from `https` and not from git

**This PR does NOT provides**
- Support for Python2 (not tested really)
- Support for Google Test

Fixes #1000 
Fixes #1155
Fixes #976

#### Full test of functionalities
1. Open a `Visual Studio x64 Native Tools Command Prompt` and navigate to the project folder
1. Remove `carla/Build` folder (if exists)
1. Run `make clean launc [... cut]
closed 2019-02-27 by nsubiron