carla-simulator/carla #1158: Better control over pedestrians

nsubiron, 2019-01-21:
Child of #630.

Allow for better control over pedestrians/walkers

- [ ] Changing characteristics like props and clothes colors from the Python API.
- [ ] Detect whether they have been run over (event? is_alive function?).

jagdishbhanushali, 2019-03-07:
Hi @nsubiron 
I am working on generating all variety of pedestrians with different props. I saw an event in unreal editor for destroying props along with destroying pedestrian. I think you must be near to finishing this feature. Please let me know much time it gonna take to finish this feature.

Thanks.

nsubiron, 2019-03-08:
Hi @jagdishbhanushali,

At the moment no one is working on this feature. We added it to our backlog because is something we would like to have, but we are currently busy with other tasks with higher priority for the Carla challenge.

JunweiLiang, 2019-07-23:
Hi @nsubiron,
Thank you for the great work! Any update/timeline on this? In v0.9.6 there are only 14 kinds of pedestrian and there is no attribute like "clothes" or "color" to change.
Thanks!

germanros1987, 2020-03-19:
The new pedestrian module should take care of this issue.

closed 2020-03-19 by germanros1987