carla-simulator/carla #1166: bash on border of road segments

Unlingius, 2019-01-23:
Hi! I observe strange effect. I have created my own map with RoadRunner and imported it to Carla. Now, when my car ride with low speed from one road segment to another, i observe "push". It looks like a microbump on the road between two segments. In some cases pitch turnes up to 1 degree. What can cause this?  

Also, i observe, that there are no push when my car has maximal acceleration forward. My version it because car's nose turned up in this case.
How to fix it? I tried to change some collision settings in ue scene, but it has not solved my problem. Thank you.

marcgpuig, 2019-01-23:
Hi @Unlingius 
Just to make sure, are you using our Unreal autopilot (that uses physics)?

Unlingius, 2019-01-23:
@marcgpuig  No. I even don't use python client now, only UE in CarlaGameMode, and the problem is observed. In client is the same.

marcgpuig, 2019-01-24:
Hi @Unlingius 
Is the geometry flat? Or maybe do you have a strange collision somewhere?
Could you provide some short video of this for further investigation?

Unlingius, 2019-01-24:
@marcgpuig  here are short video, fbx and opendrive files. There are 4 road segments in fbx. Additionally, i observed, that only 4-1 border has that effect. 
[border_push.mp4.zip](https://github.com/carla-simulator/carla/files/2791696/border_push.mp4.zip)
[scene_3.zip](https://github.com/carla-simulator/carla/files/2791702/scene_3.zip)

marcgpuig, 2019-01-25:
Hi @Unlingius

This is a problem in RoadRunner.

@XGodina use to have the same issue. A quick workaround is simply to change the number of polygons that RoadRunner generates to force it to recalculate the geometry. To do so, go to `Tools > Scene Settings` and play a bit with `Curve Tolerance` and the other parameters.

Hope it helps!

Unlingius, 2019-01-25:
@marcgpuig Thank you for your reply! Now i can't try it, because my RR trial-license has expired. Maybe later.

closed 2019-04-03 by stale[bot]