carla-simulator/carla #1171: InitSDL() failed, cannot create application instance.

Saixiaoma, 2019-01-24:
CARLA_0.9.2  UnrealEngine 4.19 Ubuntu16.04  

When I follow the tutorial to run the file  ./CarlaUE4.sh  
These are the issues I see in the log file:
[2019.01.24-10.47.37:142][  0]LogInit: Warning: Could not initialize SDL: XRandR support is required but not available
[2019.01.24-10.47.37:142][  0]LogInit: Error: FLinuxApplication::CreateLinuxApplication() : InitSDL() failed, cannot create application instance.
[2019.01.24-10.47.37:142][  0]LogLinux: FLinuxPlatformMisc::RequestExit(bForce=true, ReturnCode=1)
[2019.01.24-10.47.37:142][  0]LogLinux: FLinuxPlatformMisc::RequestExit(1)

wx9698, 2019-01-29:
Hi, I also came across this problem. Have you solved it?

Rick831022, 2019-02-13:
the same, at CARLA_0.8.2

bhaprayan, 2019-02-20:
Try installing libxrandr-dev.

skoppula, 2019-03-07:
Obtaining the same issue on CARLA_0.9.4. Unfortunately, installing libxrandr-dev did not do the trick.

bhaprayan, 2019-03-24:
What does your setup look like? I solved this issue a while back. Turns out that there's a problem with getting the OpenGL drivers to work with NVIDIA when running through VNC. The setup that worked for me is using TurboVNC (for visualizing the rendering).
If you're launching in headless mode (/wo a display), then a variant of the following command should work:
`SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -carla-server`

closed 2019-05-31 by stale[bot]

andeyeluguo, 2019-10-28:
@bhaprayan , I can run the command 
SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -carla-server
How can I  launch in head mode?
I can launch in xfce4 desktop. What should I do? Thank you very much!

wangchangquan, 2020-09-17:
i see the error when run a carla in docker,and cann't set SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0. how can i solve it .

Merealtea, 2022-08-08:
Hi, guys, I meet the same problem. And I install carla by compiling the source code, which means I have no carlaUE4.sh, so I wonder how to run this command:

SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh -carla-server