carla-simulator/carla #1177: Cross Track Error varibale

PercyJaiswal, 2019-01-27:
Hello, I am using stable version 0.8.2. pre-built linux version, and was trying to build PID controller in Carla. I was looking for help on couple of following points.

1. Is there a variable available which I can track to measure my car's distance from center of lane, often called 'cross-track error'? Right now, I am just starting at a known starting point and consider the known good x-location (when car would be at center of lane) coordinate as target value.
2. Is there a way of knowing if car has reached (or is reaching) a junction or a curve. Right now I am not aware of how to detect a junction or a curve road from game engine.

I have tried using Lane/off-road intersection value to bound my car within lane. But they dont serve as a steady / constant target value for a PID controller. coz when I am within lane, both their values are 0.

Appreciate your inputs.
Regards!

Sakrust, 2019-01-29:
This is basically all implemented in 0.9.1+ with the new waypoint API.

PercyJaiswal, 2019-01-31:
Thanks @Sakrust . I will start using 0.9.2 version.

closed 2019-01-31 by PercyJaiswal