carla-simulator/carla #1180: Failed to download content, too many users have downloaded

nsubiron, 2019-01-27:
It seems we are reaching the limit of downloads that we can do from Google Drive.

[image]

quanheng123, 2019-01-27:
Hi, just met this problem when re-set up, but what I'm now using the assets in 0.8.2, where can I find a alternative link to download? 
Thanks!

nsubiron, 2019-01-27:
I'm afraid we don't have any other link available.

mrwesleycrusher, 2019-01-27:
Would this be why when Update.sh downloads Content.tar.gz it actually fetches a "HTML document, ASCII text, with very long lines" ?

nsubiron, 2019-01-28:
Seems to be fixed for now

shivnetra1411, 2019-01-29:
While running ./Update.sh getting this error. 

Backing up existing Content...
'/home/kpit/carla/Unreal/CarlaUE4/Content/Carla' -> '/home/kpit/carla/Unreal/CarlaUE4/Content/Carla_20190129125907'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 555, in urlopen
    self._prepare_proxy(conn)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 753, in _prepare_proxy
    conn.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 252, in connect
    ssl_version=resolved_ssl_version)
  File "/usr/lib/python3/dist-packages/urllib3/util/ssl_.py", line 305, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "/usr/lib/python3.5/ssl.py", line 377, in wrap_socket
    _context=self)
  File "/usr/lib/python3.5/ssl.py", line 752, in __init__
    self.do_handshake()
  File "/usr/lib/python3.5/ssl.py", line 988, in do_handshake
    self._sslobj.do_handshake()
  File "/usr/lib/python3.5/ssl.py", line 633, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:645)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 376, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 589, in urlopen
    raise SSLError(e)
requests.packages.urllib3.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:645)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./Util/download_from_gdrive.py", line 82, in <module>
    download_file_from_google_drive(args.id, args.destination)
  File "./Util/download_from_gdrive.py", line 59, in download_file_from_google_drive
    response = session.get(url, params={'id': id}, stream=True)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 492, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 480, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 588, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 447, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:645)

xmba15, 2019-02-25:
Hi, I am trying to use ./carla/Update.sh now and it seems like the links are not downloadable now. What is the alternative way to build carla then?

xmba15, 2019-02-25:
Seems like the contents are reachable with 
```
git lfs clone https://bitbucket.org/carla-simulator/carla-content Unreal/CarlaUE4/Content/Carla
```
If that is the case, is it better to only use git lfs inside Update.sh instead of downloading from google drive?

PR #1415 (2019-03-19): Migrate Content to AWS
#### Description

Migrated Update.sh to download the Content from AWS. Download is a bit slower than Google Drive, but should solve issues related to reaching download quota limit each weekend.

If installed, it will use `aria2c` to speed up download, otherwise wget.

In parallel to this I have created a Jenkins pipeline to the Content repo that automatically uploads the package to AWS.

* Fixes #1180.
* Maybe #84?

#### Possible Drawbacks

Uhm possibly a lot, but let's try and see how it works, rolling back to Google Drive is fairly easy.

---
This change is [[image]](h [... cut]
closed 2019-03-19 by nsubiron