carla-simulator/carla #1181: sensor.other.gnss not found in Carla 0.9.2

kk2491, 2019-01-28:
Hi Guys,

I am using pre-compiled version of Carla (0.9.2 version) for development purposes. Kindly help me in understing the below 2 doubts, I found while running the simulation and agents.

1. I am not able to find the ```sensor.other.gnss``` in 0.9.2 version, I have tried fetching the ```blueprint_library```, but could not find it. Is it yet to be added in latest pre-compiled version?

2. Position of the Player vehicle and other actors. Could you please let me know what is the reference for position of player vehicle. 

Thank you,
KK

nsubiron, 2019-01-28:
Hi @kk2491,

1. The GNSS will enter in 0.9.3, the PR was merged last week, but you can download the latest nightly build from our [Jenkins server](http://158.109.9.218:8080/blue/organizations/jenkins/carla/activity?branch=master) (It's in the artifacts section on each build).
2. The reference is an arbitrary point in the map, in Town03 is the center of the roundabout.

kk2491, 2019-01-29:
@nsubiron Thanks for providing the details.

1. I have started building carla (master branch from github) from source thinking it has GNSS plugin. Now started downloading the latest build from Jenkins server. Let me check and get back to you. 
And is there a ETA on when 0.9.3 would be available in Github like other versions do.

2. Thanks, after doing some testing I was able to get the reference point and understood it is taken arbitrarily in the map.

Thank you,
KK

kk2491, 2019-01-30:
@nsubiron So tomorrow it will be released.. :-) Cool. :-) I will wait for it then. 
Thank you so much for the help.

kk2491, 2019-01-30:
@nsubiron .. One more doubt. Just wanted to know if **IMU** or **INS** plugin is available. If not could you please let me know when it would be released?

Thank you,
KK

nsubiron, 2019-01-30:
Afraid not, there are no plans on short term of adding those sensors. However, you can access velocities and accelerations of all the actors, with that you can emulate those sensors in Python.

kk2491, 2019-02-05:
@nsubiron Sorry for the late reply. Thanks for the confirmation. Let me find the ways to emulate **IMU** using the available data.

kk2491, 2019-03-01:
@nsubiron I was just going through the latest ```master``` branch and found some new Python APIs which look very interesting. However I believe the pre-built version is not released yet to try those new APIs. Could you please let me know the release date for ```0.9.4```. ??

nsubiron, 2019-03-01:
@kk2491 due date yesterday :sweat_smile:. We are still finishing last couple of things and doing some testing, the official pre-built version later today, meanwhile you can get [last nightly build from master](http://158.109.9.218:8080/blue/organizations/jenkins/carla-simulator%2Fcarla/detail/master/22/artifacts), that has already all the API functionality.

kk2491, 2019-03-01:
@nsubiron Thanks for the swift reply. Thats awesome. I will wait for the official release. :-) 
Also I am downloading the last nightly build from master.

closed 2019-05-07 by stale[bot]