carla-simulator/carla #1191: VNC CARLA Display

wx9698, 2019-01-29:
Hi,

Recently I want to use google cloud to run the carla. Because the GPU on my pc is now enough to run Carla smoothly, I try to run Carla on Google Cloud and see the game world via VNC viewer. Here is what I did, I first created a VM instance with GPU on Google Cloud and then use VNC viewer to connect this instance and work on the GUI of that instance. Then I download the compiled version of Carla. However, every time I wanted to run ./CarlaUE4.sh, the terminal always showed the error message "InitSDL() failed, cannot create application instance". Can someone solve this? Thanks!

closed 2019-04-06 by stale[bot]

knmitri, 2019-07-09:
any update on that issue ?

thefatbandit, 2020-06-23:
you can use nomachine instead of vnc. there seems to be driver problems with VNC.
https://www.nomachine.com/AR03Q01019
https://www.youtube.com/watch?v=eolNziYLe9Y