carla-simulator/carla #1193: autopilot not working in 0.9.2

rguptatx, 2019-01-30:
I am running carla on ubuntu 16.04. I compiled default version of carla and connect to it using automatic_control.py using Town3. The car turns to the right in the next 5 meters and crashes consistently everytime I run the python script.

lijecaru, 2019-02-04:
Did the script automatic_control.py worked out of the box? My script doesn't run, I'm using ubuntu 16.04 and Carla 0.9.3

xksteven, 2019-02-21:
I needed to apply this fix:
https://github.com/carla-simulator/carla/pull/1236

One line change to the file PythonAPI/agents/navigation/global_route_planner.py and then automatic_control.py works for me.

Ubuntu 16.04 
Carla 0.9.3

77hanzzi, 2019-04-09:
Same question. This script worked successfully in Town1 and Town2, but when it comes to Town3, this script always fails. Does anyone come out the possible solution?

closed 2019-06-15 by stale[bot]

Lebron-Huang, 2019-07-02:
> Same question. This script worked successfully in Town1 and Town2, but when it comes to Town3, this script always fails. Does anyone come out the possible solution?

Can you tell me how to run in town 1 and town 2？thank you