carla-simulator/carla #1194: Get the ID of the player vehicle instantly

kk2491, 2019-01-30:
Hi Guys,

I have a doubt, kindly help me in finding the correct solution. Given below is the problem statement.

Lets say I have spawned around 80 vehicles in the environment. And after that I have started the ```manual_control.py``` script to start the player vehicle.

Now if I run ```actors = world.get_actors().filter('vehicle.*')``` it is going to return all the actors along with the player vehicle. Now I want to fetch only player vehicle and monitors its position, velocity, add sensors etc.

If I am writing a python script, how would I unique identify the player vehicle out of all actors ?

For example I want to uniquely distinguish the player vehicle dynamically. 

Thank you,
KK

nsubiron, 2019-01-30:
We recently added the "role_name" attribute to actors, the player in manual control has its role set to "hero" automatically
```py
for actor in world.get_actors():
    if actor.attributes.get('role_name') == 'hero':
        player = actor
        break
```

kk2491, 2019-01-30:
@nsubiron that's awesome. Thanks for letting me know. That information would be very helpful in my future works. Will try it out and update you soon.

closed 2019-04-07 by stale[bot]