carla-simulator/carla #1195: [0.8.2] Error: CDO Constructor (CarlaMapGenerator): Failed to find StaticMesh'/Game/Static/RoadLines/St_Road_XCross_LaneMarking.St_Road_XCross_LaneMarking'

ChrisHRZ, 2019-01-30:
i use Carla 0.8.2 and UE4.18.3. And i download content 0.8.2 by myself. i extract it in /Unreal/CarlaUE4/content . And use ./Setup.sh -s and the ROOT_UE4=(my UE4 folderr) ./Rebuild.sh. It report errors as shown below.
`[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: === Handled ensure: ===
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: 
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: Ensure condition failed: !FPackageName::IsShortPackageName(Path) [File:/home/chrish/carla/UnrealEngine-4.18.3-release/Engine/Source/Runtime/CoreUObject/Private/UObject/SoftObjectPath.cpp] [Line: 45]
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: Cannot create SoftObjectPath with short package name '~'! You must pass in fully qualified package names
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: Stack: 
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2b4168cf FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2b2bccd5 FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2b41686c FLinuxPlatformStackWalk::StackWalkAndDumpEx(char*, unsigned long, int, unsigned int, void*)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2b45e202 FDebug::EnsureFailed(char const*, char const*, int, wchar_t const*)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2b45ed9f FDebug::OptionallyLogFormattedEnsureMessageReturningFalse(bool, char const*, char const*, int, wchar_t const*, ...)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2abfb688 FSoftObjectPath::SetPath(FString)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2abfc55c FSoftObjectPath::ImportTextItem(wchar_t const*&, int, UObject*, FOutputDevice*)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2aa0cf9c UScriptStruct::ImportText(wchar_t const*, void*, UObject*, int, FOutputDevice*, FString const&, bool)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2ab6e124 UStructProperty::ImportText_Internal(wchar_t const*, void*, int, UObject*, FOutputDevice*) const
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2aae1832 UObject::LoadConfig(UClass*, wchar_t const*, unsigned int, UProperty*)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2ac25529 FObjectInitializer::PostConstructInit()
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2ac24aed FObjectInitializer::~FObjectInitializer()
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2aa108a0 UClass::CreateDefaultObject()
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d1ca433bc UGameMapsSettings::GetGameDefaultMap()
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d28603b08 FURL::FURL(wchar_t const*)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d27eea6f7 UNetConnection::UNetConnection(FObjectInitializer const&)
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2aa10896 UClass::CreateDefaultObject()
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2aa105df UClass::CreateDefaultObject()
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d2ac0ad67 ProcessNewlyLoadedUObjects()
[2019.01.30-05.56.22:887][  0]LogOutputDevice: Error: [Callstack] 0x0000000000419f48 FEngineLoop::PreInit(wchar_t const*)
[2019.01.30-05.56.22:888][  0]LogOutputDevice: Error: [Callstack] 0x0000000000429be8 GuardedMain(wchar_t const*)
[2019.01.30-05.56.22:888][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d20172e34 CommonLinuxMain(int, char**, int (*)(wchar_t const*))
[2019.01.30-05.56.22:888][  0]LogOutputDevice: Error: [Callstack] 0x00007f2d1fbbe830 /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7f2d1fbbe830]
[2019.01.30-05.56.22:888][  0]LogOutputDevice: Error: [Callstack] 0x0000000000416a69 /home/chrish/carla/UnrealEngine-4.18.3-release/Engine/Binaries/Linux/UE4Editor(_start+0x29) [0x416a69]
[2019.01.30-05.56.22:888][  0]LogOutputDevice: Error: 
`
`[2019.01.30-05.56.30:235][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.01.30-05.56.30:235][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.01.30-05.56.30:236][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.01.30-05.56.30:236][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.01.30-05.56.30:237][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.01.30-05.56.30:237][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.01.30-05.56.30:238][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.01.30-05.56.30:238][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.01.30-05.56.30:239][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.01.30-05.56.30:239][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.01.30-05.56.30:239][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.01.30-05.56.30:240][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.01.30-05.56.30:240][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.01.30-05.56.30:240][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.01.30-05.56.30:241][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.01.30-05.56.30:241][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.01.30-05.56.30:242][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.01.30-05.56.30:242][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.01.30-05.56.30:243][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png' error.
[2019.01.30-05.56.30:243][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_24x.png
[2019.01.30-05.56.30:244][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.01.30-05.56.30:244][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.01.30-05.56.30:244][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png' error.
[2019.01.30-05.56.30:245][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Mac/Platform_Mac_128x.png
[2019.01.30-05.56.30:254][  0]LogStreaming: Warning: Failed to read file '../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png' error.
[2019.01.30-05.56.30:254][  0]LogSlate: Could not find file for Slate resource: ../../../Engine/Content/Editor/Slate/Launcher/Android/Platform_Android_24x.png
[... cut ...]
[2019.01.30-05.57.25:509][  0]LogModuleManager: Shutting down and abandoning module SSL (7)
[2019.01.30-05.57.25:511][  0]LogModuleManager: Shutting down and abandoning module CrashDebugHelper (4)
[2019.01.30-05.57.25:511][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2019.01.30-05.57.25:511][  0]LogExit: Exiting.
`
And it stop here. i wonder how can i fix it?
The Chinese "无法寻找到文件" means 'can't find the file'.
I compile Carla with Clang-3.9 and compile UE4 with Clang-5.0. Is that the cause of the problem?

nsubiron, 2019-02-08:
This is probably caused because there was an error downloading the asset content. Try deleting the folder `Unreal/CarlaUE4/Content` and running

    ./Update.sh

closed 2019-04-16 by stale[bot]

OAkyildiz, 2020-07-28:
Why is the path `/Game/Carla/..` and not 'Unreal/CarlaUE4/Content'? Where are these files are being copied?

MarinaMiao, 2021-01-21:
I've met this too

MarinaMiao, 2021-01-21:
I've met this too with Carla 0.8.2 and UE4.18

lovesuiyue, 2021-11-29:
Content.tar.gz    99%[==================> ]   6.01G  5.15MB/s  ,  1 second left .  ./Update.sh：54: 21796  Segment error , Core dumped , wget ${CONTENT_LINK} -O Content.tar.gz  (carla 0.9.5, UE4.21)

please help me!!!

vvnchrsto, 2022-05-21:
> Content.tar.gz 99%[==================> ] 6.01G 5.15MB/s , 1 second left . ./Update.sh：54: 21796 Segment error , Core dumped , wget ${CONTENT_LINK} -O Content.tar.gz (carla 0.9.5, UE4.21)
> 
> please help me!!!

I am stuck here as well. Can anyone point me in the right direction?