carla-simulator/carla #1199: Waypoint queries

PercyJaiswal, 2019-01-31:
Hello, I am trying to build a path planning algorithm in Carla. I am using Version 0.9.2 and was needing guidance on following points.

1. Is there a way to view waypoints on map, so that I can visually select start and end locations. Just like I could determine start and end points in version 0.8.2 using output from view_start_positions.py

2. Once I have selected waypoints in my path planning algo, is there a way I can tell my vehicle to travel between those waypoints in autopilot mode?

Appreciate your help, regards!

Sakrust, 2019-02-06:
1. You can use the debug functions provided to draw points on your map in Carla simulator view for every waypoint selected by recommended_spawn_points or use the no rendering mode to display the start positions in an bird's eye view.

2. Use the Agent classes provided with the PythonAPI to follow a list of waypoints.

closed 2019-04-14 by stale[bot]

Derekabc, 2019-05-27:
@Sakrust do you have a sample for navigating a car by a list of waypoints (.txt file)?