carla-simulator/carla #1203: Spawning walkers using PythonAPI Carla 0.9.2

mallela, 2019-01-31:
Hello,

I am using Carla 0.9.2 on Ubuntu 16.04. I want to spawn pedestrians. I noticed that I can spawn vehicle blueprints using the PythonAPI. I looked for walker/pedestrian blueprints in the blueprint list and they weren't listed. Is it possible to access walker blueprints? If yes, how?

Should this be input using CarlaSetting.ini file or is carlasetting.ini depricated in Carla 0.9.2?

Thanks,
Praneeta

dixantmittal, 2019-02-04:
Hi Praneeta,

You can get the blueprint for walker by using `blueprint_library.filter('walker.*')` in ver 0.9.3. Not sure about ver 0.9.2 though.

-Dixant

closed 2019-02-11 by mallela

Curry30h, 2019-03-07:
> Hi Praneeta,
> 
> You can get the blueprint for walker by using `blueprint_library.filter('walker.*')` in ver 0.9.3. Not sure about ver 0.9.2 though.
> 
> -Dixant

How do you add pedestrians in the virtual environment in CARLA? Can you tell me more about it? I copied the code of adding a car to add pedestrians in spawn_npc, but it is useless. I also use 0.9. 3 version, I hope you can teach me.Thank you！

dixantmittal, 2019-03-07:
Hi @Curry30h, 
In Carla, pedestrian is just another "Actor".  So, to add a pedestrian, you need a blueprint of the _Pedestrian_ actor, which you can find from Blueprint Library by using `blueprint_library.filter('walker.*')`. After getting the blueprint, spawn the actor by `world.spawn(blueprint, location)`.

A sample code for your reference:
`pedestrian = world.spawn_actor(random.choice(blueprint_library.filter('walker*')), Transform())`