carla-simulator/carla #1205: Multi-agent RGB sensor problem

hongyuli, 2019-02-01:
Hi, I'm using Carla 0.9.2 on Ubuntu and got some problems if multiple vehicles are facing the same direction while recording RGB images.

Specifically, I put 4 vehicles which are close to each other,  facing the same direction, and all recording RGB, depth and segmentation data using the save_to_disk function. The logged depth and segmentation images are fine, but many of the RGB images either cannot be opened or messed up. When saving to png files, the images cannot be opened with 'Fatal error reading PNG image file: IDAT: CRC error'.

1. Firstly I thought this might be my system's problem when save_to_disk calling lower level APIs. Thus, I reinstall and upgrade libglib of my Ubuntu, since this package may cause the problem of saving png files. But this does not work. Besides, saving depth and segmentation png files are totally fine, I guess this is not the problem.

2. I thought saving the image to other formats might be a workaround. However, if I save the images to jpeg and tiff, even wired things happened that some images messed up with other views.
[image]
[image]

Let me upload my logging client code here for anyone wants to reproduce the issue.

Thanks!
[HL_Logging.zip](https://github.com/carla-simulator/carla/files/2820044/HL_Logging.zip)

nsubiron, 2019-02-08:
I had similar issues and eventually I found out that the problem was that I was saving the images simultaneously to the same file path. I was reusing a Python variable in a for loop while setting the `listen` callback, and since Python lambdas capture variables by reference I ended up saving all the images to the same path.

hongyuli, 2019-02-08:
Got it. You are right, adding the sensor id to the file path fixed the problem.

Thanks a lot.

closed 2019-02-08 by hongyuli