carla-simulator/carla #1207: Carla 0.9.2 gets slower after every iteration of scenario_runner

anshulpaigwar, 2019-02-01:
I am using Scenario_runner with Carla 0.9.2. I observed while doing multiple repetitions Carla gets slower and slower. I am running Carla in benchmark mode with fps = 10hz and after every repetition framerate drops (frame rate =  3 Hz after 50 reps). I am destroying all the actors after every rep but it seems memory is not completely released. When I restart the Carla server the frame rate goes back to normal.

### Is there any way to refresh/reset Carla world using python API ?? 
`world.destroy()` doesn't solve the problem.

nsubiron, 2019-02-08:
Yes, that looks like if some sensor is not properly destroyed. 

> Is there any way to refresh/reset Carla world using python API ??

Not yet, but soon.

MikeZLin, 2019-03-01:
Try launching the simulator and the Scenario runner using python sub-processes, to restart the server kill the process and relaunch. Steps to launch and kill a process can be found here:
https://stackoverflow.com/questions/4789837/how-to-terminate-a-python-subprocess-launched-with-shell-true

anshulpaigwar, 2019-03-06:
This the hack I also tried. I made a shell script to launch and kill the server in loop

closed 2020-06-26 by germanros1987