carla-simulator/carla #1208: Why cross product in path selection?

silgon, 2019-02-01:
I have one question regarding `basic_agent.py` of the Python API.  I would like to know why you are using the cross product to decide where to turn in a junction when you have multiple choices. 

You can see the part of the code that I'm talking about in the next link: https://github.com/carla-simulator/carla/blob/master/PythonAPI/agents/navigation/basic_agent.py#L101-L111

I ask the question because in the case of the robot in position (0,0) the decision of the road will be  always the first choice in the iteration. Am I mistaken?

Thanks!

pravinblaze, 2019-02-08:
I am not sure if I understand the question correctly but assuming that you are thinking the co ordinates of the robot is what is used the make the vector, let me try to explain.

Say at an intersection you have encountered more than one choice of waypoints representing different paths,
We construct unit vectors along the current heading of the robot and the direction towards path choices. In this situation, we compare the cross products to determine which path is to the left and which is to the right.

silgon, 2019-02-20:
Sorry for the delay. I realized you're finding the sine of the angle by using the cross product and the unitary vector of each direction. Based on your code and the following equation.

[image]
(Image extracted from wikipedia)

Thanks for the explanation.

closed 2019-02-20 by silgon