carla-simulator/carla #1209: question about Routeplanner.

quanheng123, 2019-02-02:
Hi, 
I have a question about routeplanner, here i want to create massive of routeplanners in editor, with massive of coordination points  data which for the locations of triggervolume , splinepoints and so on , is there any loluntion that i can automatically genrate these routeplanners ?   
thanks.

quanheng123, 2019-02-03:
Here is what I doing, I just personal modified the routplanner file to give them a fixed location and route and write a python scripts to generate massive of cpp and h just changing location data  in every cpp and h file, just like this 

`#include "Carla.h"
#include "MyRoutPlanner0.h"

#include "Util/RandomEngine.h"
#include "Vehicle/CarlaWheeledVehicle.h"
#include "Vehicle/WheeledVehicleAIController.h"

#include "Engine/CollisionProfile.h"

static AWheeledVehicleAIController * GetVehicleController10 (AActor *Actor)
{
  auto *Vehicle = (Actor->IsPendingKill() ? nullptr : Cast<ACarlaWheeledVehicle>(Actor));
  return (Vehicle != nullptr ?
      Cast<AWheeledVehicleAIController>(Vehicle->GetController()) :
      nullptr);
}

static const USplineComponent *PickARoute10 (
    URandomEngine &RandomEngine,
    const TArray<USplineComponent *> &Routes,
    const TArray<float> &Probabilities)
{
   check(Routes.Num() > 0);

  auto Index = RandomEngine.GetIntWithWeight(Probabilities);
  check((Index >= 0) && (Index < Routes.Num()));
  return Routes[Index];
}

// Sets default values
AMyRoutPlanner0 :: AMyRoutPlanner0 (const FObjectInitializer& ObjectInitializer) :
  Super(ObjectInitializer)
{	RootComponent = ObjectInitializer.CreateDefaultSubobject<USceneComponent>(this, TEXT("SceneRootComponent"));
  RootComponent->SetMobility(EComponentMobility::Static);
  TriggerVolume0 = CreateDefaultSubobject<UBoxComponent>(TEXT("TriggerVolume0"));
  TriggerVolume0->SetupAttachment(RootComponent);
  TriggerVolume0->SetHiddenInGame(true);
  TriggerVolume0->SetMobility(EComponentMobility::Static);
  TriggerVolume0->SetCollisionProfileName(FName("OverlapAll"));
  TriggerVolume0->bGenerateOverlapEvents = true;
  TriggerVolume0->SetBoxExtent(FVector(200, 200, 200),false);
  TriggerVolume0->SetWorldLocation(FVector(96447.39179266617, 57826.342385751195, 124.28526651032428));
  TriggerVolume0->SetWorldRotation(FRotator(0, 357, 0));

  RouteDirection0 = CreateDefaultSubobject<USplineComponent>(TEXT("RouteDirection0"));
  RouteDirection0->SetupAttachment(RootComponent);
  RouteDirection0->SetHiddenInGame(true);
  RouteDirection0->SetMobility(EComponentMobility::Static);
  RouteDirection0->RegisterComponent();
  RouteDirection0->SetWorldLocationAtSplinePoint(0,FVector(96447.39179266617, 57826.342385751195, 124.28526651032428));
  RouteDirection0->SetWorldLocationAtSplinePoint(1,FVector(96450.87560461834, 57749.46527731372, 124.05791036702851));
  RouteDirection0->AddSplineWorldPoint(FVector(96641.5511756204, 53731.94724371424, 111.406857482271));
  Routes.Add(RouteDirection0);

  
}

#if WITH_EDITOR		
void  AMyRoutPlanner0 ::PostEditChangeProperty(FPropertyChangedEvent& PropertyChangedEvent)
{
  

  

  Super::PostEditChangeProperty(PropertyChangedEvent);
  const auto Size = Routes.Num();
  if (PropertyChangedEvent.Property && (Size != Probabilities.Num()))
   {
[... 293 lines cut ...]
                                        ^~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/llvm-5.0/lib/clang/5.0.1/include/stddef.h:120:24: note: expanded from
      macro 'offsetof'
#define offsetof(t, d) __builtin_offsetof(t, d)
                       ^~~~~~~~~~~~~~~~~~~~~~~~
/home/heng/carla/Unreal/CarlaUE4/Plugins/Carla/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/Carla/TrafficSignBase.gen.cpp:192:220: note: 
      insert an explicit cast to silence this issue
  ...1, nullptr, STRUCT_OFFSET(ATrafficSignBase, TrafficSignState), Z_Constru...
                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                 static_cast<int32>(                              )
/home/heng/UnrealEngine_4.18/Engine/Source/Runtime/Core/Public/Templates/UnrealTemplate.h:143:40: note: 
      expanded from macro 'STRUCT_OFFSET'
#define STRUCT_OFFSET( struc, member )  offsetof(struc, member)
                                        ^~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/llvm-5.0/lib/clang/5.0.1/include/stddef.h:120:24: note: expanded from
      macro 'offsetof'
#define offsetof(t, d) __builtin_offsetof(t, d)
                       ^~~~~~~~~~~~~~~~~~~~~~~~
8 errors generated.
`

Why there is a error with relation of traffic light and traffic sign ?  I dont get it....
then I just try to rebuild just with fisrt 12 files, I always succssed.
How can I fix this ?
Any help will be appriciated!

closed 2019-04-11 by stale[bot]

candytaco, 2020-12-04:
I know this over a year and a half old, but commenting anyways for posterity in case anyone else runs into this issue. I am also working on some custom code. It is, as the first compiler error indicates, a unknown type issue. All the macro expansion errors are red herrings (which led me to waste two good weeks). You need to add an explicit `#include` for `UTrafficSignComponent.h`.

I found the solution here: https://answers.unrealengine.com/questions/730971/unknown-type-name-utextrendercomponent-while-follo.html.

My hunch as to why this happens is the Unreal build tools, which fills in the `UFUNCTION`, `UPROPERTY`, etc, macros before the code is compiled by the regular C++ compiler. I found that if I commented some very specific `UFUNCTION` macros out (before I included the explicit `#include`), the code would compile successfully. Perhaps, depending on where Unreal macros are, the include order of headers get shuffled up in the autogenerated files. But that's just a hunch. I guess, to be on the safe side, it's good practice to have explicit `#includes` for all the classes used in each file, given the Include What You Use model.