carla-simulator/carla #1210: Crash on 0.9.3 compiled version

overlapjho, 2019-02-02:
I get this error when I ```./CarlaUE4.sh``` on Ubuntu 16.04 and Carla 0.9.3 compiled version

```
[2019.02.02-13.28.35:288][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapRoad requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapRoad
[2019.02.02-13.28.35:288][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.02.02-13.28.35:289][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalkCurb requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalkCurb
[2019.02.02-13.28.35:289][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.02.02-13.28.35:289][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalk_00 requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalk_00
[2019.02.02-13.28.35:289][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.02.02-13.28.35:290][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapLaneMarking requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapLaneMarking
[2019.02.02-13.28.35:290][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.02.02-13.28.35:291][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Road_N2 requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Road_N2
[2019.02.02-13.28.35:291][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.02.02-13.28.35:291][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Concrete requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Concrete
[2019.02.02-13.28.35:291][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.02.02-13.28.35:292][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SideWalks/SidewalkN4/WetPavement_SidewalkN4 requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SideWalks/SidewalkN4/WetPavement_SidewalkN4
[2019.02.02-13.28.35:292][  0]LogStreaming: Error: Found 0 dependent packages...
[2019.02.02-13.28.35:293][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/LaneMarking/Lanemarking requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/LaneMarking/Lanemarking
[2019.02.02-13.28.35:293][  0]LogStreaming: Error: Found 0 dependent packages...

[2019.02.02-13.28.52:422][  0]LogRHI: Error: Failed to link program. Current total programs: 132 binary bytes: 405281624
  log:
error: vertex shader output `out_TEXCOORD0' specifies noperspective interpolation qualifier, but fragment shader input specifies no interpolation qualifier

Failed to find symbol file, expected location:
"/home/overlapjho/carla/CarlaUE4/Binaries/Linux/CarlaUE4.sym"
Fatal error: [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLShaders.cpp] [Line: 3718] 
Failed to link program. Current total programs: 132, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554```

overlapjho, 2019-02-03:
this only happend whe it has ```SDL_VIDEODRIVER=offscreen``` I am trying to run with autoware

sbastas, 2019-02-04:
I have the same issue with mesa 18.2.8 drivers(Works on 0.9.2). With the latest amdgpu-pro drivers(18.40) CARLA runs fine but the textures are mainly black/missing.

wendersonj, 2019-02-06:
Having the same issue here.

wendersonj, 2019-02-06:
Having the same issue on Ubuntu 18.04LTS.

overlapjho, 2019-02-07:
hey guys, its seems the problem came from graphic drivers, i recently bought 1050ti and install gpu drivers then the problem is gone. This error might happen without gpu

wendersonj, 2019-02-07:
I thought so. But I already had the drivers installed. The issues was that I didn't **restarted my pc after I installed**. Silly, right? Haha. Anyway, I have a GF830M in my notebook using the driver 410.78 for Linux. 

Thanks for the tip @overlapjho !

riemannulus, 2019-02-11:
Hey, I got same problem and solved. My problem is without execute [Nvidia Optimus](https://wiki.archlinux.org/index.php/NVIDIA_Optimus). If you using carla on laptop, execute with **primus ./CarlaUE4.sh**
It work on me.

BrendanSaltafomaggio, 2019-03-14:
If someone could comment or update this with a proper workaround, that would be awesome. On 2 different machines I tried updating the drivers and rebooting after installation, but to no avail.
both systems are Ubuntu 16.04
One has a radeon amd gpu and the other has a stock intel.

zlw21gxy, 2019-03-18:
[image]
[image]

I face the same problem......does anyone know what happened...I am going to be crazy

zhefan, 2019-03-22:
Same issue on Ubuntu 16.04.6 with 1080 Ti (driver version: 418.43). I am running CARLA 0.9.4.

Johnnyswong, 2019-04-04:
Maybe this issue is related to the OpenGL library? Try 

`sudo apt install x11-xserver-utils libxrandr-dev`

bernatx, 2019-04-29:
Hi,

Anybody with that problem? I have checked a bit and seems the problem could be using the flag -opengl3 in the command line. It seems that Unreal Engine for Linux can not use that version, and then shaders can not compile and program can not link properly. So, try the command line without that flag and see again, for example with: ./CarlaUE4.sh Game/Carla/Maps/Town01 -benchmark -fps=10

Regards

jxj801259, 2019-04-30:
I face the same problem

bernatx, 2019-04-30:
Hi,

It happens using the opengl3 parameter?
Can you give me some more details about how to try to reproduce that problem?

closed 2019-04-30 by jmarrr

pawel-ziecina, 2019-06-11:
on my laptop I also observing this issue with official carla v0.9.5 release:

```
[2019.06.11-09.31.59:118][  0]LogRHI: Error: Failed to link program. Current total programs: 134 binary bytes: 409286582
  log:
error: vertex shader output `out_TEXCOORD0' specifies noperspective interpolation qualifier, but fragment shader input specifies no interpolation qualifier

Failed to find symbol file, expected location:
"/tmp/carla-20190400-095-original/CarlaUE4/Binaries/Linux/CarlaUE4.sym"
Fatal error: [File:/var/lib/jenkins/UnrealEngine_4.21/Engine/Source/Runtime/OpenGLDrv/Private/OpenGLShaders.cpp] [Line: 3718] 
Failed to link program. Current total programs: 134, precompile: 0
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=68112 LargeMemoryPoolOffset=199248 
Malloc Size=128368 LargeMemoryPoolOffset=327632 
Malloc Size=501903 LargeMemoryPoolOffset=829551 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

- Ubuntu 16.04 on laptop with intel GPU
- observable also on 
- `glxinfo -B` output:
```
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Intel Open Source Technology Center (0x8086)
    Device: Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2)  (0x591b)
    Version: 18.0.5
    Accelerated: yes
    Video memory: 3072MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 3.0
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) HD Graphics 630 (Kaby Lake GT2) 
OpenGL core profile version string: 4.5 (Core Profile) Mesa 18.0.5
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 3.0 Mesa 18.0.5
OpenGL shading language version string: 1.30
OpenGL context flags: (none)

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 18.0.5
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
```

pawel-ziecina, 2019-06-11:
Issue confirmed also on 2 other laptops (same brand and model as above; Ubuntu 16.04 and 18.04)

pawel-ziecina, 2019-06-11:
Let's move discussion to #1627

Derekabc, 2019-06-20:
@jmarrr how do you solve this problem?

sound-design, 2020-01-30:
Same problem with 0.9.5 pre-compiled.

qhaas, 2021-03-21:
Seeing this (sometimes) with CARLA 0.9.8 using opengl on a Tesla V100 in an Ubuntu 18.04 environment:
```
Failed to find symbol file, expected location:
"CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.sym"
4.23.1-0+++UE4+Release-4.23 517 0
Disabling core dumps.
CommonUnixCrashHandler: Signal=11
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
```

QianwenZhu, 2021-06-21:
> Seeing this (sometimes) with CARLA 0.9.8 using opengl on a Tesla V100 in an Ubuntu 18.04 environment:
>
> ```
> Failed to find symbol file, expected location: [...]

Have you solved this problem? the same problem as you， I used precompiled CARLA 0.9.8.

libing64, 2022-01-15:
`./CarlaUE4.sh 
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Failed to find symbol file, expected location:
"/opt/carla-simulator/CarlaUE4/Binaries/Linux/CarlaUE4-Linux-Shipping.sym"
LowLevelFatalError [File:Unknown] [Line: 2089] 
Out of memory on Vulkan; MemoryTypeIndex=9, AllocSize=512.000MB
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
`
Ubuntu18.04 Carla-0.9.13

ilkunviktor, 2022-08-16:
Hello, I got almost same error. And I started to receive it 100%, when I set a breakpoint into macro (several lines macro in our project). When I remove the bug is gone:

 Signal 5 caught.
 Malloc Size=65538 LargeMemoryPoolOffset=65554 
 CommonUnixCrashHandler: Signal=5
 [2022.08.15-16.06.35:408][  0]LogCore: === Critical error: ===
 Unhandled Exception: SIGTRAP: trace trap
 
 [2022.08.15-16.06.35:408][  0]LogCore: Fatal error!