carla-simulator/carla #1211: Automatic_control Carla 0.9.3

lijecaru, 2019-02-02:
I'm in Ubuntu 16.04 x64 and just downloaded Carla 0.9.3. I want to run the file automatic_control.py but an error appears saying that it couldn't import agents.navigation.roaming_agent. Actually a red line indicating error appears when I also import Carla, but I didn't pay attention to it, since the manual_control.py script works. I wonder if I'm missing something. Thanks.
[image]

moh3th1, 2019-02-03:
I have the same error on Ubuntu 18.04 
[image]

``` 
/home/user/anaconda3/bin/python /home/user/Downloads/CARLA_0.9.3/automatic_control.py
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/home/user/Downloads/CARLA_0.9.3/automatic_control.py", line 80, in <module>
    from agents.navigation.roaming_agent import *
ModuleNotFoundError: No module named 'agents'
``` 
But I tried to edit the imports and it worked for me
 Replace every ``` .agents. ``` with ``` PythonAPI.agents. ```

lijecaru, 2019-02-04:
> I am facing the same issue. For some reasons, @moh3th1 solution did not work for me. But adding 'PythonAPI' to `sys.path` worked.
> 
> [image]
> 
> Add the below to the code,
> 
> ```python
> try:  
> 	sys.path.append('PythonAPI')   
> except IndexError:  
> 	pass
> ```

lijecaru, 2019-02-04:
> I am facing the same issue. For some reasons, @moh3th1 solution did not work for me. But adding 'PythonAPI' to `sys.path` worked.
> 
> [image]
> 
> Add the below to the code,
> 
> ```python
> try:  
> 	sys.path.append('PythonAPI')   
> except IndexError:  
> 	pass
> ```

I tried both solutions and it didn't work. The solution provided by @moh3th1 gives me also an import error:
ImportError: No module named PythonAPI.agents.navigation.roaming_agent

And the solution from @vvrs provides the following output:

pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 127.0.0.1:2000

    Example of automatic vehicle control from client side.

ERROR: Node 244 not reachable from 205
Traceback (most recent call last):
  File "automatic_control.py", line 702, in main
    game_loop(args)
  File "automatic_control.py", line 631, in game_loop
    spawn_point.location.z))
  File "PythonAPI/agents/navigation/basic_agent.py", line 62, in set_destination
    route = grp.plan_route((x1, y1), (x2, y2))
  File "PythonAPI/agents/navigation/global_route_planner.py", line 55, in plan_route
    route = self.path_search(origin, destination)
  File "PythonAPI/agents/navigation/global_route_planner.py", line 115, in path_search
    weight='length')
  File "<decorator-gen-260>", line 2, in astar_path
  File "/home/linda/anaconda2/lib/python2.7/site-packages/networkx/utils/decorators.py", line 73, in _not_implemented_for
    return not_implement_for_func(*args, **kwargs)
  File "/home/linda/anaconda2/lib/python2.7/site-packages/networkx/algorithms/shortest_paths/astar.py", line 134, in astar_path
    raise nx.NetworkXNoPath("Node %s not reachable from %s" % (source, target))
NetworkXNoPath: Node 244 not reachable from 205
WARNING: attempting to destroy an actor that is already dead: Actor 104 (vehicle.lincoln.mkz2017)
Destroying ego-vehicle!

Do you maybe have other ideas of what the problem could be? Thank you!

YashBansod, 2019-02-11:
For the problems regarding import error, its best to add the following:
```python
try:
    sys.path.append(glob.glob('PythonAPI')[0])
except IndexError:
    print("PythonAPI package wasn't found")

try:
    sys.path.append(glob.glob('**/agents')[0])
except IndexError:
    print("PythonAPI/agents package wasn't found")

try:
    sys.path.append(glob.glob('**/agents/navigation')[0])
except IndexError:
    print("PythonAPI/agents/navigation package wasn't found")

try:
    sys.path.append(glob.glob('**/agents/tools')[0])
except IndexError:
    print("PythonAPI/agents/tools package wasn't found")
```

Also make sure that the pip package **networkx** is installed.
However @lijecaru 's problem is completely unrelated to these.

closed 2019-02-14 by nsubiron

geekhut, 2019-02-21:
> > I am facing the same issue. For some reasons, @moh3th1 solution did not work for me. But adding 'PythonAPI' to `sys.path` worked.
> > [image]
> > Add the below to the code,
> > ```python
> > try:  
> > 	sys.path.append('PythonAPI')   
> > except IndexError:  
> > 	pass
> > ```
> 
> I tried both solutions and it didn't work. The solution provided by @moh3th1 gives me also an import error:
> ImportError: No module named PythonAPI.agents.navigation.roaming_agent
> 
> And the solution from @vvrs provides the following output:
> 
> pygame 1.9.4
> Hello from the pygame community. https://www.pygame.org/contribute.html
> INFO: listening to server 127.0.0.1:2000
> 
> ```
> Example of automatic vehicle control from client side.
> ```
> 
> ERROR: Node 244 not reachable from 205
> Traceback (most recent call last):
> File "automatic_control.py", line 702, in main
> game_loop(args)
> File "automatic_control.py", line 631, in game_loop
> spawn_point.location.z))
> File "PythonAPI/agents/navigation/basic_agent.py", line 62, in set_destination
> route = grp.plan_route((x1, y1), (x2, y2))
> File "PythonAPI/agents/navigation/global_route_planner.py", line 55, in plan_route
> route = self.path_search(origin, destination)
> File "PythonAPI/agents/navigation/global_route_planner.py", line 115, in path_search
> weight='length')
> File "", line 2, in astar_path
> File "/home/linda/anaconda2/lib/python2.7/site-packages/networkx/utils/decorators.py", line 73, in _not_implemented_for
> return not_implement_for_func(*args, **kwargs)
> File "/home/linda/anaconda2/lib/python2.7/site-packages/networkx/algorithms/shortest_paths/astar.py", line 134, in astar_path
> raise nx.NetworkXNoPath("Node %s not reachable from %s" % (source, target))
> NetworkXNoPath: Node 244 not reachable from 205
> WARNING: attempting to destroy an actor that is already dead: Actor 104 (vehicle.lincoln.mkz2017)
> Destroying ego-vehicle!
> 
> Do you maybe have other ideas of what the problem could be? Thank you!
@lijecaru 
I encountered the same problem > NetworkXNoPath: Node 244 not reachable from 205,
have you managed to solve this?

lijecaru, 2019-02-22:
Hi @geekhut , I installed networkx using `pip install networkx` and then I restarted my PC. After this it worked, hope it helps!

tsc1989, 2019-09-05:
For windows this will help.
install package using pip utility as the following 

go to directory PythonAPI/carla/agents 
and create setup.py file and write the following code in that file

from distutils.core import setup
setup(name='agents',
    version='0.9.5', #doesn't matter I guess
    py_modules=['agents'],
    )

Now go to anaconda terminal and do cd to directory PythonAPI/carla
and apply following command to install agents

pip install -e agents

This will install agents and resolve the agents.navigation not found issue

Mecalzmy, 2020-02-06:
> Hi @geekhut , I installed networkx using `pip install networkx` and then I restarted my PC. After this it worked, hope it helps!

I installed networkx using `pip install networkx` and then I restarted my PC.  But it still didn't works.

Lala341, 2021-05-17:
It's works for me.
`try:
    sys.path.append(os.path.dirname(os.path.dirname(os.path.abspath(__file__))) + '/carla')
except IndexError:
    pass`

PR #1248 (2019-02-14): fix for 1211

#### Description

Small fix for 1211 

Fixes #1211 

#### Where has this been tested?

  * **Platform(s):**  Ubunty 16.04
  * **Python version(s):** 3.5.5
  * **Unreal Engine version(s):** NA

#### Possible Drawbacks

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/1248)