carla-simulator/carla #1219: Carla Virtual reality

moh3th1, 2019-02-04:
Is it possible to compile carla for VR headset and how ?

XGodina, 2019-02-06:
Hi, @moh3th1 . We haven't done CARLA thinking about that possibility. But we don't discard doing in the future because is very interesting. If you want to try it and share is welcome.

shriram17, 2019-02-15:
Hey, I managed to get Carla 0.8.3 working in VR mode on Windows using the UE4Editor, here are the steps I followed 

1. Add the suitable VR plugin to UE4 `Plugins>SteamVR/OculusVR`  depending on your device, restart the UE4Editor
2. Execute the following commands via editor or blueprint, to switch to VR mode
`r.VSync 0 , vr.bEnableHMD 1, vr.bEnableStereo 1, r.SetRes 1920x1080f`
3. Setup the Player or Spectator camera to desired location and test using the VR Preview mode that can be found in the dropdown under Play button in th editor. 
Note : You may need to add "HeadMountedDisplay" module in the PrivateModule Section of Carla.Build.cs file, if you wish to run HMD module code from inside the Carla Plugin.

XGodina, 2019-02-18:
Hi @shriram17, 
Thanks for share with us how to use VR with Carla. We haven't worked yet with this technology. But we have it in mind for our plans for the future.

closed 2019-04-26 by stale[bot]

soulslicer, 2019-09-11:
> Hey, I managed to get Carla 0.8.3 working in VR mode on Windows using the UE4Editor, here are the steps I followed
>
> 1. Add the suitable VR plugin to UE4 `Plugins>SteamVR/OculusVR`  depending on your [...]

Could you please give a more detailed set of instructions? Does this work at an arbitrary camera that we define in carla as well?

cjerome0321, 2019-09-23:
> Hey, I managed to get Carla 0.8.3 working in VR mode on Windows using the UE4Editor, here are the steps I followed
>
> 1. Add the suitable VR plugin to UE4 `Plugins>SteamVR/OculusVR`  depending on your [...]

Hi @shriram17 ,

I am also trying to get VR working. Could you please tell me if you can manually control a vehicle/pedestrain using VR in your set up? 

Thanks!

Harsharma2308, 2020-02-26:
Can someone guide to a script or some comments or more development on this? Thanks.

filippogandolfi, 2020-05-14:
Is there some new stuff or suggestion for the VR mode?
I'm try to following @shriram17 but it's not so clear

uservinu, 2021-05-27:
Hello all, 
I managed get VR working in CARLA 0.9.10 on windows with HTC Vive. You have to add a VR camera pawn or attach it to some actor. see https://docs.unrealengine.com/4.26/en-US/SharingAndReleasing/XRDevelopment/VR/SteamVR/HowTo/StandingCamera/ for details.
But the performance(frame rate is less than 90hz, effect similar to screen tearing is happening) is not so great with my Nvidia GTX1080.  Please share how to improve the performance if somebody has experience in this.
Thanks,
Vinu

AhmedAredah, 2021-11-03:
> Thanks,

So where to include the VRCamera pawn? do you mean we need to alter the source code?

AhmedAredah, 2021-11-03:
> Hello all,
> I managed get VR working in CARLA 0.9.10 on windows with HTC Vive. You have to add a VR camera pawn or attach it to some actor. see [...]

So where to include the VRCamera pawn? do you mean we need to alter the source code?

ashishgx, 2021-11-19:
has anyone tried this with Oculus?

uservinu, 2021-12-03:
@AhmedAredah I am attaching the VRCameraPawn to a Vehicle. See the BP of the vehicle 
[image]

Piti98, 2022-08-19:
For everyone interested, i wrote a step by step guide on how to add VR to the Carla Unreal Editor, see https://github.com/carla-simulator/carla/discussions/5667