carla-simulator/carla #1220: Get list of actors from running carla map[in autopilot mode]

Dhairyashil-kpit, 2019-02-05:
Dear all, 

I am new to opensource world and trying to test carla python API, so I wrote the sandbox for getting carla actors from the current running map. so I found following observations.

1. In main() when I remove the while and simply called the get_actors() some time it returns the empty list.
2. In main() when I call the get_actors() in while , in the beginning it return the empty list, then it return the actors. 

Questions:
 1. I am trying to query the actors from running carla map so my approach is right ? 
 2.  How to do asynchronous call using carla python api.
 3. In manual and autopilot mode does Carla give different output. 

`
"""
Carla actors properties.
"""
import logging
import random
import time 

import carla
from carla_settings import CarlaSettings
from get_carla_client import GetCarlaClient

class GetActorsFromWorld(object):
    '''
    returns all properties of carla world.
    '''
    def __init__(self, carla_client):
        '''
        initilize the carla client.
        '''
        self.carla_client = carla_client
        self.carla_logger = logging.getLogger('carla_client_logger')
        self.carla_logger.info("Carla Server initialized....")
        self.world = None
        self.actors_list = None
        self.actor_dict = dict()
    
    def get_dict_of_actors(self):
        '''
        return available dict of actors obj.
        '''
        self.world = self.carla_client.get_world()                
        self.actors_list = self.world.get_actors()
        self.actor_dict["vehicle"] = self.actors_list.filter('vehicle.*')
        self.actor_dict["traffic"] = self.actors_list.filter('traffic.*')        
        print(self.actors_list.filter('vehicle.*'))
        #return self.actor_dict

def main():
    '''
    main calling.
    '''    
    carla_client = GetCarlaClient(CarlaSettings.carla_host, CarlaSettings.carla_port)
    actors_obj = GetActorsFromWorld(carla_client.get_client())
    # Get continues list of actors.
    # 
    while(True):
        carla_actors = actors_obj.get_dict_of_actors()
        time.sleep(1)    

if __name__ == "__main__":
    main()`

Dhairyashil-kpit, 2019-02-05:
I used:
` # Wait for the world to be ready`
`self.world.wait_for_tick(10.0)` 
 Now whenever I run the sandbox it gives me actors when world is ready.
working fine.

Sakrust, 2019-02-06:
I don't exactly understand your question to be honest, but information about the API is provided in their documentation: https://carla.readthedocs.io/en/latest/python_api/

Dhairyashil-kpit, 2019-02-06:
Hi @Sakrust, 
Thank you for your reply.
Sorry for complicated post.

I am trying to do following:
1. Using pythonAPI query the all actors from running carla server [Done. refer above script]
2. Trying control carla vehicle like apply break, etc. [ running in _manual_control.py_ in autopilot mode ] via different client script _control_carla_vehicle.py_ , But when carla server is running in autopilot mode it doesn't allow us to override the properties/controls/attributes of vehicle. so how to do it?
3.

nsubiron, 2019-02-08:
Hi @Dhairyashil-kpit,

> 1. Using pythonAPI query the all actors from running carla server

Yes, you are doing it right. The list of actors is updated each tick in the background, if you retrieve the list before the first update you may get an empty list. I think we can fix this inside Carla module, but meanwhile the best approach is calling `world.wait_for_tick()` first time before `world.get_actors()`.

> 2. Trying control carla vehicle like apply break, etc. [running in _manual_control.py_ in autopilot mode]

The problem here is that both the autopilot controller and your script are sending controls to the same vehicle, so most likely the autopilot is overwriting your commands every tick. I'm afraid at this point your only choice is to disable autopilot.

I'm actually working now on fixing these issues by ensuring that the user commands are applied last every tick. Nevertheless, it's very likely the autopilot would fail if you deviate too much of the route by overwriting its commands.

nsubiron, 2019-02-15:
#1257 fixes `world.get_actors()` returning an empty list.

aaronhma, 2019-04-20:
I think this issue is still in the latest version of CARLA.

closed 2019-06-26 by stale[bot]