carla-simulator/carla #1222: How to change towns in Carla 0.9.3

imran514, 2019-02-06:
I have downloaded Linux compiled version of Carla 0.9.3, and i explore new towns that are added in this new version

Sakrust, 2019-02-06:
Use /Game/Carla/Maps/<MapName> as an argument when launching your carla binaries.

dixantmittal, 2019-02-11:
Hi @imran514, As you have downloaded Linux compiled version, simply run the command `$ ./CarlaUE4.sh TownXX`. I noticed that now you don't need to add prefix /Game/Carla/Maps/TownXX to load TownXX.

stevendae, 2019-03-20:
How do I change towns in the editor?

Sakrust, 2019-03-20:
Navigate to Carla/Maps and double click on the map you wish to load.

kimiya66, 2019-04-02:
> Hi @imran514, As you have downloaded Linux compiled version, simply run the command `$ ./CarlaUE4.sh TownXX`. I noticed that now you don't need to add prefix /Game/Carla/Maps/TownXX to load TownXX.

Thank you :)

closed 2019-06-08 by stale[bot]

Miknow78, 2022-01-05:
Anyone knows how to set a map as default when open the simulator?