carla-simulator/carla #1224: Is it possible to set two cameras in the same location to call the callback function in the same frame? [0.9.3]

nikhilmohan33, 2019-02-07:
I am currently writing callback functions for each sensor I am attaching to my actor.  I want to collect RGB/segmentation/depth data from the same location and capture all three synchronously. Currently the frame number for each of my separate cameras is a couple off when I store into disk. I am assuming this is because each image is being stored on a different tick of the server. 

Is it possible to get all three images synchronously (captured at the same instant)? I looked at #1042, but I dont think this thread addressed capturing images at the same instant.

Thank you.

Unlingius, 2019-02-07:
Hi! If you don't need realtime speed, at least you may use idea from https://github.com/carla-simulator/carla/issues/1032

nikhilmohan33, 2019-02-07:
Thank you for the quick response. From #1032 how did you manage to create the next set of images offline? Did you simply use the track data to place a vehicle there with a different sensor? I am looking to use 1280x720 images. 
I have multiple GPUs so compute does not seem to be a problem.

Unlingius, 2019-02-08:
@nikhilmohan33 yes, I deactivated physics and set different sensor to stored location. It is possible to spawn only sensor, without vehicle. But one problem i faced is there are no ways to know if "set new position" is already applied. Workaround for me was wait for receiving two or three images and take the the last. It's not beautiful, but works.

closed 2019-04-16 by stale[bot]