carla-simulator/carla #1225: Carla 0.9.3 : Sensor's Parent Information

krdindorkar19, 2019-02-07:
Hi,

In manual_control.py, when we spawn collision sensor, I added this code below it : 

>>>        self.sensor = world.spawn_actor(bp, carla.Transform(), attach_to=self._parent)
>>>        sensor_list=world.get_actors().filter("sensor.*")
>>>        for sensor in sensor_list:
>>>            print(sensor.type_id)
>>>            print(sensor.parent)
>>>        print("self.sensor.parent ============>",self.sensor.parent)

I am getting output as : 

sensor.lidar.ray_cast 
None
sensor.lidar.ray_cast 
None
sensor.lidar.ray_cast 
None
sensor.lidar.ray_cast 
None
sensor.other.collision 
None
self.sensor.parent ============> Actor(id=79, type=vehicle.bh.crossbike)

So sensors we get from world.get_actors() is different from actual spawned sensors ?

krdindorkar19, 2019-02-07:
As per scenario_runner, 
In manual_control.py sensors are spawned on ego vehicle. However whenever the sensor data is needed, sensors are re-spawned on ego vehicle.

Is it possible to access sensor data from the sensors which are already spawned on ego vehicle through manual_control.py?

Dasdude, 2019-02-07:
Apparently only server side sensor object does parent field is None. You can check it by checking parent field when you attach the sensor instance to vehicle on client as opposed to retrieve the sensor object from server. if you have access to your client side sensor object you can get the parent.

krdindorkar19, 2019-02-08:
I have spawned camera sensor in client 1 : 
code : 
print("Camera : ",world.camera_manager.sensor)
output : 
Camera :  Actor(id=380, type=sensor.camera.rgb)

From client 2 I am trying to retrieve it : 
>def get_dict_of_actors(self):
>    '''
>    return available dict of actors obj.
>    '''
>    # Wait for the world to be ready          
>    self.world.wait_for_tick(10.0)
>    self.actors_list = self.world.get_actors()
>    self.actor_dictionary["sensor"] = self.actors_list.filter('sensor.*')        
>    return self.actor_dictionary
>
>def sensor_listener(args):
>    actors_obj = GetActorsFromWorld(carla_world)	
>    carla_actors = actors_obj.get_dict_of_actors()
>    if carla_actors['sensor']:
>        camera=carla_actors['sensor'].filter('sensor.camera.rgb')
>        for actor in camera:
>            print("Camera : ",actor,actor.parent)	

Output of client_2 : 
>Camera :  Actor(id=199, type=sensor.camera.rgb) None
>Camera :  Actor(id=200, type=sensor.camera.rgb) None
>Camera :  Actor(id=201, type=sensor.camera.rgb) None
>Camera :  Actor(id=202, type=sensor.camera.rgb) None
>Camera :  Actor(id=209, type=sensor.camera.rgb) None
>Camera :  Actor(id=210, type=sensor.camera.rgb) None
>Camera :  Actor(id=216, type=sensor.camera.rgb) None
>Camera :  Actor(id=251, type=sensor.camera.rgb) None
>Camera :  Actor(id=252, type=sensor.camera.rgb) None
>Camera :  Actor(id=253, type=sensor.camera.rgb) None
>Camera :  Actor(id=191, type=sensor.camera.rgb) None
>Camera :  Actor(id=171, type=sensor.camera.rgb) None
>Camera :  Actor(id=177, type=sensor.camera.rgb) None
>Camera :  Actor(id=178, type=sensor.camera.rgb) None
>Camera :  Actor(id=241, type=sensor.camera.rgb) None
>Camera :  Actor(id=380, type=sensor.camera.rgb) None
>Camera :  Actor(id=184, type=sensor.camera.rgb) None
>Camera :  Actor(id=242, type=sensor.camera.rgb) None
>Camera :  Actor(id=185, type=sensor.camera.rgb) None
>Camera :  Actor(id=192, type=sensor.camera.rgb) None
>Camera :  Actor(id=193, type=sensor.camera.rgb) None
>Camera :  Actor(id=203, type=sensor.camera.rgb) None

So for camera id 380 also it is giving parent as None

krdindorkar19, 2019-02-08:
[get_sensor_parent.txt](https://github.com/carla-simulator/carla/files/2844476/get_sensor_parent.txt)

I have run manual_control.py(Client 1) which is printing spawned camera's id and get_sensor_parent.py (Client 2).

nsubiron, 2019-02-08:
The actor retrieved from `world.get_actors()` is a different instance (in Python) than the one retrieved from `world.spawn_actor(...)`, it should contain the same information about the actor, but it's possible that something went wrong and the parent got missing.

Marking this as bug to investigate more.

nsubiron, 2019-06-17:
Fixed on #1744.

closed 2019-06-17 by nsubiron